paparazzi/paparazzi #1445: Realtime Plotter: Average and standard dev of curves not shown anymore

flixr, 2015-11-29:
At least under Ubuntu 14.04 with Unity the average and std deviation values of each curve are not shown any more (in the curves menu).
They are properly computed and the update callback is executed, but it seems that the value label which is set as an image is not displayed:
```
let stdev_value = GMisc.label ~text:"N/A" () in
let _item = submenu_fact#add_image_item ~image:stdev_value#coerce ~label:"Stdev" () in
```
So even before the first value is calculated it should show `N/A`, but that is not the case...

OpenUAS, 2015-12-03:
So I've noticed testing low throughput link, though somehow it was just me, good to know it is not :)
Used Xubuntu 14.04. Although it does show N/A in my case but never a value...

flixr, 2015-12-03:
Well, the update of the menu (and hence display of the values) was disabled with #1447...
With which version were you testing? Before or after that?
If the display of the images and values works under Xubuntu 14.04 (before it was disabled), can you post the used lablgtk, gtk, etc. versions?

flixr, 2015-12-03:
Also this has nothing to do with low throughput on link....

OpenUAS, 2015-12-09:
Will investigate and post feedback and used versions and OS version etc. etc. 
I like the nice deviation feature of the realtime logger a lot when debugging live datafeeds, so one more thing on my list...

podhrmic, 2016-07-06:
@OpenUAS any updates here? @flixr maybe we can remove the avg/std-dev buttons from the plotter since they are not working? Although it would be nice to have them, the manpower is probably more needed elsewhere.

podhrmic, 2016-12-28:
This would be useful, but since the avg/stddev buttons are already removed we probably should just close this.

flixr, 2016-12-28:
Closing this, please open a new feature request if this functionality should be added again...

closed 2016-12-28 by flixr