paparazzi/paparazzi #1451: No gps: no problem!

rmeertens, 2015-12-01:
Hey everybody, 

With the rise of autonomous indoor drones that avoid  obstacles and can have an indication of their own speed I would like to enable the NAV mode of paparazzi to control such drones. 
Currently using NAV mode is impossible indoors as you can't get a GPS fix, and without a gps position you can't start your drone. 

@dewagter and I circumvent this with ugly hacks, but I would like your opinion on a mergable solution. 

I see two options: 
- create a define in your airframe file like: "NO_GPS_NEEDED_FOR_NAV"
- create a "gps_none" that always says you have a 3d fix. 

Please let me know what your opinion is on how to implement this neatly.

gautierhattenberger, 2015-12-01:
I think the system should not care if it has GPS or not, but rather get the information from the state interface that position/speed/attitude is valid or not (and not only initialized).
This is something I wanted to do for a long time but didn't find the time.

flixr, 2015-12-01:
I agree, this needs to be handled in a nicer way... especially if you have a GPS but can actually fly without a fix.
What should work for now: don't add a gps subsystem and change the `GpsIsLost` define to `#define GpsIsLost() FALSE` in https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/autopilot.c#L52

rmeertens, 2015-12-01:
So for now, how about this? You can add a define to your airframe file: 
#if USE_GPS
#include "subsystems/gps.h"
#else
#if DONT_NEED_GPS
#define GpsIsLost() FALSE
#else
#define GpsIsLost() TRUE
#endif
#endif

flixr, 2015-12-01:
Jep, I think that should work. I even like your first suggestion better: `NO_GPS_NEEDED_FOR_NAV`
If you can confirm that this is all that is currently needed I'm happy to add this.

commit 42939370 (2015-12-01): [rotorcraft] add NO_GPS_NEEDED_FOR_NAV

If a GPS is not used, and you want to use NAV mode, add to your airframe file:
<define name="NO_GPS_NEEDED_FOR_NAV" value="TRUE"/>
and make sure that you don't have a gps subsystem added.

This should close #1451 for now.
A better way to deal with this is probably still needed (e.g. if you want to be able to have a GPS connected, but not use it indoors).
closed 2015-12-01 by flixr