paparazzi/paparazzi #1458: Getting more done in GitHub with ZenHub

flixr, 2015-12-02:
Hola! @flixr has created a [ZenHub](http://www.zenhub.io) account for the **paparazzi** organization. ZenHub is the leading team collaboration and project management solution built for GitHub.

----

#### How do I use ZenHub?

To get set up with ZenHub, all you have to do is **[download the browser extension](http://www.zenhub.io)** and log in with your GitHub account. Once you do, you’ll get access to ZenHub’s complete feature-set immediately.

#### What can ZenHub do?

ZenHub adds a series of enhancements directly inside the GitHub UI:

- Real-time, customizable task boards for GitHub issues;
- Burndown charts, estimates, and velocity tracking based on GitHub Milestones;
- Personal to-do lists and task prioritization;
- “+1” button for GitHub issues and comments;
- Drag-and-drop file sharing;
- Time-saving shortcuts like a quick repo switcher.

### [Add ZenHub to GitHub](https://www.zenhub.io)

_Still curious? See [more ZenHub features](http://www.zenhub.io/features) or read [user reviews](https://chrome.google.com/webstore/detail/zenhub-for-github/ogcgkffhplmphkaahpmffcafajaocjbd/reviews). This issue was written by your friendly ZenHub bot, posted by request from @flixr._

[image]

flixr, 2015-12-02:
Uh, didn't expect this to happen...

OpenUAS, 2015-12-09:
would be good...

closed 2016-03-11 by flixr