paparazzi/paparazzi #1459: UBLOX 8 autoconfiguration

dewagter, 2015-12-03:
- NAV5 message has bits used in a different way

dewagter, 2015-12-07:
Direct link: https://www.u-blox.com/sites/default/files/products/documents/u-bloxM8_ReceiverDescrProtSpec_%28UBX-13003221%29_Public.pdf

flixr, 2015-12-07:
According to the 7to8 migration guide the UBX-CFG-NAV5 message should be backwards compatible: https://www.u-blox.com/sites/default/files/products/documents/ubx7toM8-SWMigrationGuide_AppNote_%28UBX-13003254%29.pdf

gautierhattenberger, 2015-12-07:
That's why I don't understand why it is not set correctly. Or it is a side effect of an other issue.

flixr, 2015-12-07:
No idea... and I don't have a M8 to test...

tenyan, 2015-12-11:
Use NAV-PVT for ubx7+ modules , use NAV-SOL, NAV-POSLLH, NAV-VELNED and NAV-TIMEUTC for ubx6?

flixr, 2015-12-13:
NAV-PVT doesn't really give you anything extra that you don't already get with NAV-SOL, NAV-POSLLH and NAV-VELNED, which are all still available for ublox 7 and 8...

flixr, 2015-12-13:
@dewagter do you have any more insights on why the UBX-CFG-NAV5 doesn't seem to be working as expected?

gautierhattenberger, 2015-12-15:
The reason it is not working (at least one of the reasons...) is that the autobaud function is not working. Or exactly it is not very robust to properly catch the ACK. And if it happens to work (either it found the correct value while testing them or it falls back to the default and it is the correct one), it will not set the uart to the "new" value. It will send the message to set the baudrate to the last tested rate. This is because `UBX_GPS_BAUD` is defined to `GPS_LINK.baudrate` which is set to the last value, not the initial one.
The end result in most cases is that the modules are saturated with messages at 9600 bauds and that the reply of the version request fails, and then the NAV config is sent instead of NAV5.
I'm going to try this patch:
- save target baudrate at init
- use NAV5 by default if version couldn't be fetched
Any other suggestions ?

closed 2015-12-18 by flixr

dewagter, 2015-12-22:
Working for me now in master.

PR #1480 (2015-12-18): [gps] fix ubx ucenter autobaud and autoconf
should fix #1459 
tested on UBX M8M module with default config at startup