paparazzi/paparazzi #1463: [Feature request] Inter-vehicle position tracking for rotorcraft

kirkscheper, 2015-12-06:
I am currently working (in my free time) on swarming with some quadrotors and would like each member of the swarm to keep track of the other members so they can preform some separation assurance. I saw that in fixed wing there is some capability to do this typically called traffic_info and tracks vehicles using ac_info msgs. I however haven't found anything like it for rotorcraft.

A couple of questions:
Was this functionality used much in fixed wing?
Is there anything already implemented in rotorcraft to do this?
Is this interesting for others to have this functionality in rotorcraft?
Is it alright to extend the the tracking from ac_info msgs to remote_gps and/or remote_GPS_small msgs (the last one is especially interesting due to its small size).

Any advice or comments are welcome (especially from @flixr or @fvantienen)

gautierhattenberger, 2015-12-06:
Hi,

Some short answers:

> A couple of questions:
> Was this functionality used much in fixed wing?
Not much I think, but these multi-UAV was part of a research project 
that led to really flights with collision avoidance in 2010.
>
> Is there anything already implemented in rotorcraft to do this?
Not yet

> Is this interesting for others to have this functionality in rotorcraft?
This is something that was already asked and discussed pretty often. And 
yes it is a major interest but the main issue is that while fixed-wing 
are flying using (global) UTM coordinates, rotorcrafts are flying in LTP 
coordinates that are local to each aircraft. They could share some 
global coordinates (LLA, ECEF,...) and convert in their own frame. Just 
nobody took the time to do it properly.
>
> Is it alright to extend the the tracking from ac_info msgs to 
> remote_gps and/or remote_GPS_small msgs (the last one is especially 
> interesting due to its small size).
Not sure it helps much to mix message functionality, but you can still 
build new compact message based on the same principal.
>
> Any advice or comments are welcome (especially from @flixr 
> <https://github.com/flixr> or @fvantienen <https://github.com/fvantienen>)

flixr, 2015-12-07:
Basically Gautier mentioned said it already:
we should either add a `ACINFO_ECEF` or `ACINFO_LLA` message (lat/lon in fixed point 10e7deg).
ECEF should take less CPU to convert to NED, LLA is probably more common (not sure that matters here).

kirkscheper, 2015-12-07:
Ok, Based on this I think I will go down the road of lla for communicating the position but replace the utm format in the ac_info_ struct with a ecef format. (this will end up being a big push).

I have hit my first hurdle, the ac_info message is defined in the uplink class of messages. I assume that previously all locations were relayed to the aircraft via the ground station but I am interested in having the vehicles send the position directly to each other without the ground station in the middle. Making copies of messages in each class sounds a bit unnecessary, the option of having the aircraft send datalink messages to each other is also not ideal as the messages can also be received by the ground station and are then not correctly received as it thinks it can only receive downlink messages. 

As far as I can see there is no check at the moment if a received message is of the correct length at all. It assumes that all received messages are datalink messages but as far as I can see that is not necessarily true. You could have a correctly formatted message of a similar id of a different class. I think this is a bit dangerous...

Is there currently anyway to identify which class the message is from? If not I could check if the size of the message received. If it is not the correct size of a datalink message it is not parsed and is sent onwards to the modules who can maybe pick up and use the message, regardless of class.

gautierhattenberger, 2015-12-07:
The message system as not been design to work with air-to-air messages, that is indeed a fact. I had some students working on a new protocol some years ago to address these issues, but we couldn't finish it.
If you look at the file sw/airborne/firmwares/fixedwing/datalink.c there are some lines commented at the beginning of the dl_parse function that was suppose to parse telemetry message when the sender id is not 0 (the ground station id). But actually their are a few other problems:

- you could have several aircraft on different modems or on different channels
- when using XBee modems in API mode, downlink messages are not broadcasted and will be only received by the ground

In these cases, using the ground as relay is the only solution. And making a system that nicely adapts in all cases (without an explicit air-to-air message class and a class id in the protocol header) is a big headache.

kirkscheper, 2015-12-08:
Ah thanks!

So in summary, for air-air messages I need to use telemetry class msgs which are intercepted by the sender if in the datalink.c. Any messages which need to be relayed by the ground station will still work but will need to be datalink messages. 

I think that is fair enough.

BTW, do you want to keep the utm axis system at all for the acinfo. I have already rewritten everything for ecef format in my local branch :)

Thanks for pointing me in the right direction, really helps to know what has already been done and what the issues really are. 

If your interested, you can track the changes at https://github.com/kirkscheper/paparazzi/tree/RSSIlog

OpenUAS, 2015-12-09:
@kirkscheper thx, this would also help in Rotorcraft-Fixedwing unifying codebase. Will discuss face to face

gautierhattenberger, 2015-12-10:
UTM has some interesting properties: it is a global position (compared to LTP) but still Cartesian (compared to LLA) with nice axis (compared to ECEF) and not sensitive to initialization point (compared to ECEF/LLA/LTP combination, even if borders are a bit annoying some times). And it can be used directly with geoid altitude (easier to compare to SRTM elevation map).
Also, fixed-wing are also capable of flying over long distances where LTP shouldn't be used.

So rather than unifying position representations where it not always makes sense, I would prefer to make an (optional) LLA based navigation, where distances computations require more CPU but is still fine on newer boards (especially with stm32f4).

OpenUAS, 2015-12-23:
@gautierhattenberger , yes that sounds great to me, a very nice in depth short overview, you nailed the issue quite well, Thx!

kirkscheper, 2016-01-06:
Check #1509 for the first pull request for this

kirkscheper, 2016-02-10:
Hey all, so I am struggling with the reference frame to use between the drones and I need a bit of help. The problem points a bit to waht @gautierhattenberger mentioned above on the 10th of Dec. 

Currently in Paparazzi what was done is the position of the other aircraft is received in UTM and stored as the UTM position relative to the origin of the flight plan (UTM-NAV_UTM_XXXX0). This value was then compared directly to the ENU coordinates from the aircraft's state. This may be convenient and 'correct' when you are close to the origin but not so much when you move away and certainly not when you move to a new zone.

Additionally, most of the computations with UTM assume you are not changing zones eg. utm_of_lla_f doesn't compute the zone, it just trusts the zone input to the function. Also the macros ENU_OF_UTM_DIFF, NED_OF_UTM_DIFF, UTM_OF_ENU_ADD, UTM_OF_NED_ADD assume that the computation won't cause you to traverse zones. I am not an expert in reference frames but it doesn't seem very robust. Only utm_int_from_gps recomputes the zone if set to 0.

While testing I also noticed that onboard the altitude in the utm position is above the geoid when the ground station relays the gps position using the ac_info message the altitude is hmsl which is also not great...

Any advice you all have on this issue would be GREATLY appreciated :)

Thanks in advance for all the help.

In short, should I go ahead with the utm and try to improve all these issues or focus on lla representation which seems to have more support in pprz?

flixr, 2016-02-10:
That is actually a good but a bit tricky question...

Continuing to use UTM relative to the flight plan (as opposed to "real" UTM coordinates) is probably not a good idea in any case. At the very least we should switch to proper UTM coordinates here.
Regarding the zone transitions: how we currently handle this is that if you are on a UTM zone border or get a coordinate in the next zone, the "origin" zone is basically extended to cover that as well.

Using LLA is more general, but it will have to be converted to a local representation for distance calculation.
Also LLA conversions are comparatively costly (which might impact the older LPC based autopilots), but I think that shouldn't prevent us from using LLA here.

One important thing to keep in mind with LLA is that you shouldn't store them as floats as you loose too much precision at higher latitudes/longitudes. So we should use either doubles or the int32_t e-7deg representation.
The main pain point here is that we currently don't have a LLA conversion function that doesn't intermediately use floats, so we would need to implement one that either uses doubles or can directly use the int32 e-7deg representation without loss of precision.

kirkscheper, 2016-04-18:
@flixr, I wrote a version of the bluegiga firmware to run on the ardrone using a bluetooth dongle (+cdc-acm driver and uploaded using the python ardrone tools) allowing for (secondary) datalink over bluetooth. Which folder should I put the files?

I currently manually build the program but I can imagine that I could automate the build and upload process into pprz.

flixr, 2016-04-18:
@kirkscheper let's discuss this on gitter or an appropriate pull request...

kirkscheper, 2016-06-20:
Closed with #1638

closed 2016-06-20 by kirkscheper