paparazzi/paparazzi #1475: convert subsystems to modules

flixr, 2015-12-12:
The goal is to use the same xml description format we have for modules for subsystems (or rather convert subsystems to modules).
This is sort of a meta issue to track progress of the various sub issues...

This will yield the following advantages:
- Better documentation
- auto settings (#23)
- possibility to use periodic and event functions that are automatically called
- hook into datalink parsing
- make dependencies between subsystems more clear/explicit

With #1471 we are on track to do that, still TODO:
- [ ] possibility to have firmware specific parts (extra defines, etc), e.g. add firmware attribute to makefile section of module
- [ ] improve dependency system as per #940
- [ ] actually convert the subsystem to modules

Related issues to also keep in mind:
- #656
- #246

flixr, 2016-04-19:
Elaboration on periodic/event function usage:
Quite a few subsystems have periodic and event functions that are called from main and have/need some sort of #ifdef protection: imu, ins, gps...
But not all implementations actually need these periodic and event functions.
Converting those to modules let's you easily add those only for the cases where needed and cleans up main.

gautierhattenberger, 2021-03-25:
I consider this solved, except for moving the code from susbsystems folder to modules folder

closed 2021-03-25 by gautierhattenberger