paparazzi/paparazzi #1477: GCS Strip Buttons and Flight Blocks Non-Functional on Mac OS X

hooperfly, 2015-12-13:
The GCS strip buttons along as well as double clicking on flight blocks to trigger them no longer functions under Mac OS X. The launch, kill, resurrect, altitude and lateral shift buttons function as expected. This is behavior specific to Mac OS X. These same tests/actions under Linux 14.04 work as expected.

flixr, 2015-12-13:
Can we lump all these El Capitan issues together, since they all seem to be related?

hooperfly, 2015-12-13:
It seems odd that some of the buttons work and the strip buttons do not. Seems inconsistent and makes me wonder about the GTK+ library.

flixr, 2015-12-13:
But it was working before El Capitan, right?
And this is just one of the many "regressions" we see with lablgtk/GTK2 under El Capitan.

hooperfly, 2015-12-13:
Not a bug. User error with respect to `link` tool arguments. Had mistakenly used the `-udp_broadcast` argument by mistake. The invocation should have been just the following:

```
$PAPARAZZI_HOME/sw/ground_segment/tmtc/link -udp 
```

closed 2015-12-13 by hooperfly