paparazzi/paparazzi #1483: Flying without gps in nav mode

rmeertens, 2015-12-16:
Although I defined  NO_GPS_NEEDED_FOR_NAV I still can't seem to enter NAV mode with my joystick. 
The problem might be resolved at this line (655) in the autopilot.c file:

    /* don't enter NAV mode if GPS is lost (this also prevents mode oscillations) */
    if (!(new_autopilot_mode == AP_MODE_NAV && GpsIsLost())) {

flixr, 2015-12-19:
If ` NO_GPS_NEEDED_FOR_NAV ` is defined AND you don't have a GPS subsystem (`USE_GPS` is false) then `GpsIsLost()` is also FALSE, see https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/autopilot.c#L53

So it must be something else...

rmeertens, 2015-12-20:
Pretty sure both are defined correctly. Will continue searching this week.

flixr, 2015-12-20:
Currently I can't see where the problem would be, but If you can share your exact conf, I'll try to look into it as soon as I find the time...

OpenUAS, 2015-12-23:
* Look at your generated files in var/aircraft/ they migh give a hint the issue is. 
* A pointer to the used airframe and conf so we can reproduce the issue welcomed.

flixr, 2015-12-27:
@rmeertens are you using a GPS subsystem or not?

closed 2016-02-07 by rmeertens