paparazzi/paparazzi #1484: Message or Message Class for Passing Feature Specific Data

hooperfly, 2015-12-17:
The feature in this case is the triggering/setting of lighting data and passing it to all/each aircraft via Telemetry uplink. For example, sending RGB values to an on-board lighting system using a telemetry message/technique. Currently there seems to be at least three options for achieving this goal:

1. Use the `COMMANDS_RAW` message to send 8-bits of data to each aircraft. 
```
 <message name="COMMANDS_RAW" id="12" link="forwarded">
   <field name="ac_id" type="uint8"/>
   <field name="commands" type="int8[]"/>
 </message>
```
Given the three 8-bit values for each RGB state, that's three messages per aircraft. Barring the use of a reduced resolution space(i.e. make the light firmware only respond to data that is contained in a single byte), the overhead of packing up 8-bits to send `N` messages to `M` aircraft seems inefficient.

2. Create a new message type that 

a.) incorporates a large payload for overloading, so to speak, or
```
 <message name="COMMAND_OVERLOAD_RAW" id="12" link="forwarded">
   <field name="ac_id" type="uint8"/>
   <field name="feaure_encapsulated_message_type" type="int8[]"/>
  <field name="payload_1"    type="int32" unit="cm"></field>
  <field name="payload_2"   type="uint32" unit="ms"></field>
  ....
 </message>
```
 b.) is specific to lighting related commands.
```
 <message name="LED_CONTROL" id="<UNUSED_MESSAGE_ID>" link="forwarded">
   <field name="ac_id" type="uint8"/>
   <field name="rgb_value" type="uint32"/>
   <field name="on_off_frequency" type="uint16"/>
   ....
 </message>
```

3. Use a `MSG_CLASS` to  define a set of lighting/LED control messages:

```
  <msg_class name="LED CONTROL"> <!-- data for led light control -->
    <message name="COLOR" id="1">
      <field name="rgb_vlaue" type="unit32"/>
    </message>
    <message name="ON_OFF_FREQ" id="2">
      <field name="on_off_frequency" type="uint16"/>
    </message>
   ....
  </msg_class>
```

Not that the details of the message content or data sizes are arbitrary at this point for the purpose of discussion.

I reviewed the current message space and didn't detect other possibilities for using/overloading existing telemetry constructs. In general, overloading messages is not very maintainable in the long haul.

Any thoughts or input on how I might go about solving this issue is appreciated. Effectively I want to be able to use the ivy bus to trigger telemetry messages that an onboard system will consume and take the appropriate action.

OpenUAS, 2015-12-23:
Nothing wrong with using your own messages.xml after all it is still a CONFiguration file where you can be your own boss. So whatever works for you is fine. Also creating an own IVY agent is actually using IVY where it was designed for. You have specific wish, just create a solution that fits you best, no need to adhere to whatever PPRZ, it is opensource and extendable. Even creating a module would be an option. If you want it to even the stuff to end up in master, just a messages.xml that suits you in conf/HOOPERFLY/messages.xml and one then yous has to symlink it to use it, all is fine.

Or maybe chop up the COMMANDS_RAW into 0 = the blue end and 255 = the full RED end and move in between. for the rest You lose a bit of colorspace resolution but maybe it is not even noticable in your real LED solution.

It is great to adhere to PPRZ defaults, but IMHO we rather see something refreshing and amazing, do not feel intimidated, just go, create and.. have fun with it :)

dewagter, 2016-01-12:
In the context of paparazzi I would vote for option 2A: a generic PAYLOAD_CMD->bytes[] message to be overloaded by any struct you like, as is done with the PAYLOAD downlink message in several modules. This means you need to know the version of paparazzi and conf in order to know what the payload message means, but this is no different from settings and flightplan blocks, so I guess this must be acceptable.

Maybe it is even possible to do "sub-telemetry over telemetry", where a new set of payload messages are filled in the single master PAYLOAD message. If a sub-telemetry file were made where committed messages have low ID's (1,2,3..) and all personal ID's (not in master) start with the highest numbers (255,254,...) that might finally solve the need many people have to add telemetry messages without either having to add useless messages to master or having hopeless merging problems afterwards.

podhrmic, 2017-08-26:
@hooperfly is this still an issue for you? Have you looked at Pprzlink 2.0 if it fulfills your need?

podhrmic, 2018-04-14:
Closing for timeout.

closed 2018-04-14 by podhrmic