paparazzi/paparazzi #1495: Python tools not executing on Mac installation

hooperfly, 2015-12-21:
The python tools used in `Paparazzi Center` as well as from the command line are not executing and complain about libraries not being found. Possibly the Mac OS X install is not including all of the necessary `python` libraries required by Paparazzi? NOTE: tools work as expected under Ubuntu.

Here's the type of output displayed when trying to invoke one of the python tools:

```
$ ./sw/ground_segment/python/guided_mode_example.py 
Traceback (most recent call last):
  File "./guided_mode_example.py", line 13, in <module>
    from ivy_msg_interface import IvyMessagesInterface
  File "/Users/richardaburton/paparazzi/sw/lib/python/ivy_msg_interface.py", line 3, in <module>
    from ivy.std_api import *
ImportError: No module named ivy.std_api
```

flixr, 2015-12-22:
Do you have ivy-python installed?
At least it is not found...

hooperfly, 2015-12-22:
@flixr my guess is that the Paparazzi UAV Mac OS X install process is deficient with respect to the needed python libraries used by the various python scripts. I wanted to open an issue to capture what those might be and capture them in this thread. The `ivy-python` is indeed missing at this point.

hooperfly, 2015-12-22:
@flixr I am now able to run the script on Mac OS X but it has not affect on the simulation like the Ubuntu version. My guess, without checking, is it's related to ivy bus ip address/port differences between Ubuntu and Mac OS X.

hooperfly, 2015-12-22:
@flixr Based on my suspicion of ivy bus ip address/port differences for why the script execution has no impact on the simulation with respect to Mac OS X, the file ./sw/lib/python/ivy_msg_interface.py might be the culprit. The initialization of the ivy bus doesn't seem to specify an ip address. Thoughts?

flixr, 2015-12-22:
I am still surprised that IVY itself has no working defaults...

commit 34e076e2 (2015-12-22): [python] add pprz_env with usable default IVY_BUS

- pprz_env.IVY_BUS is either the env variable IVY_BUS or appropriate default (checks for OSX)
- some fixes for udp_link

should fix #1495 and the python part of #204
closed 2015-12-22 by flixr