paparazzi/paparazzi #1503: [conf] Group together conf files

podhrmic, 2015-12-29:
Would be possible to ground the conf files into one folder?

Right now in `conf` there are various folders that are not related to the actual paparazzi code, such as `AGGIEIAR`, `CDW` etc.

The conf files used to be together with the airframe files, but since that was not desirable for some reason, can the folders be grouped into one? For example in a new folder `aiframe_config`:
```
conf/aiframe_config/AGGIAIR
conf/aiframe_config/CDW
conf/aiframe_config/...
```

Right now it is really cluttering the `conf` directory, so it is harder to find the folder you are looking for, and it is making comparisons between branches more difficult. I could image that it is also confusing for users.

Thanks for consideration.

flixr, 2015-12-29:
I'm not quite following... what exactly do you want to put in the `airframe_config` folder and what keep in `airframes`?

podhrmic, 2015-12-29:
`airframe_config` ->The uppercase folders that now contain the conf.xml files that are selected with the start.py script. (for example AGGIEAIR, CDW...)

`airframes` -> keep it as is

That way there will be only two folders in `conf` that are related to user airframes, rather than 10 (and possibly growing).

Does it make sense now?

flixr, 2015-12-29:
hm... to me it does not make more sense to have some airframe files in conf/airframes (like the examples) and some in conf/airframe_conf ("user specific"), especially since quite some conf.xml have airframe files from examples and from user specific dirs...

If people agree that changing this helps, OK... personally I find it even more confusing to look for airframe files in conf/airframes and conf/aircframe_conf...
If the uppercase USER_FILES vs. examples is not distinction enough, I would rather have a subdir under conf/airframes where all user airframes go (e.g. conf/airframes/USERS/AGGIEAIR ) or so...

Also the user files are often more "related to actual paparazzi code" than the examples, because they are test flown more...

podhrmic, 2015-12-29:
I see - I guess what I wanted to say was that the uppercase directories in `conf` folder are distracting. If they are in lets say `conf/airframes` instead that is totally cool, because then you don't have to see them unless you want to see some specific user files.

For me as a developer it is confusing because now instead of having clearly named directories in `conf` there are now mixed other unrelated folders with user files.

So I would advocate for having as few folders in `conf` as possible, like it used to be before (for example https://github.com/paparazzi/paparazzi/tree/v5.6/conf )

flixr, 2015-12-29:
Oh, sorry.... I thought you meant the conf/airframes/AGGIEAIR directories and not the conf/AGGIEAIR... my bad, didn't read closely enough.

Yeah, I find the `conf/USER` dirs kind of annoying as well.... I liked it better when the user conf.xml was also in the subfolder of the user airframe files. (Yeah, I know conf.xml is not an airframe file... whatever).

In any case the confs were moved by @OpenUAS in https://github.com/paparazzi/paparazzi/pull/1383

You guys hash it out what you want to have....

flixr, 2016-01-12:
@OpenUAS any comments here?

OpenUAS, 2016-02-09:
Yes, I'll clean up more over time and listen to all inputs, see Gitter Developers and this. Next version will be more convenient.  @podhrmic FYI coming from PPRZ v2.x I know what was the initial conf Idea but it as "Misused" for more. Will discuss it and together we find a great solution.

podhrmic, 2016-06-21:
Is this still an issue or should we consider it resolved? Imho it works quite nicely as is now in master. @OpenUAS do you have feedback on this?

podhrmic, 2016-07-06:
@flixr Imho this issue can be closed as resolved - nobody had any complaints with the way it is now as far as I know.

closed 2016-07-06 by flixr