paparazzi/paparazzi #1518: ocamlfind: Package `glibivy' not found

datomnurdin, 2016-01-24:
It seems I got an error message when i trying to make Paparazzi. The error message like this

```
------------------------------------------------------------
Building Paparazzi version v5.6.0_stable-31-g11796a6
------------------------------------------------------------
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C data/maps
-----------------------------------------------
DOWNLOAD: google maps version code
Updated google maps version to 194
-----------------------------------------------
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C sw/lib/ocaml
OC fig.mli
ocamlfind: Package `glibivy' not found
make[1]: *** [fig.cmi] Error 2
make: *** [libpprz] Error 2
```

Please advice. Thank you.

flixr, 2016-01-24:
You are missing the OCaml ivy package, which probably should have been installed via opam.

Please have a look at http://wiki.paparazziuav.org/wiki/Installation/MacOSX.

Also the mailing list is a more appropriate place to ask for help with this as there are more OSX users who could chip in.
http://wiki.paparazziuav.org/wiki/Contact

flixr, 2016-01-28:
@datomnurdin were you able to fix your issue following the instructions on the wiki?

datomnurdin, 2016-01-28:
I will try it first.

flixr, 2016-02-27:
@datomnurdin were you able to resolve the issue by following the wiki instructions?
Can we close this issue?

closed 2016-03-11 by flixr