paparazzi/paparazzi #1545: Back up message box very unclear

kevindehecker, 2016-02-23:
[image]

At seemingly random moments, I'm confronted with this backup dialog during startup of pprz Center, which I do not want to see nor read because:
1. It is totally irrelevant at that moment. (trying to fly here!)
2. it's unclear what I'm supposed to do... even if I read it. 
3. The most persuasive and save looking button (restore auto backup), actually destroys last made settings.

Suggest to remove that dialog, and handle this without user attention. Modern GUI auto-saves without requiring user attention, but automatically creates a back-up of the previous state in case the user did not want to save.

gautierhattenberger, 2016-02-23:
Hum, I almost sure this discussion already happen somewhere...
1. I agree
2. I don't totally agree
3. Actually reading all the lines tells you that by restoring the backup you will lose the latest manual changes

Yet, I agree this is not the ideal way to do this.

But:
- unless someone wants to do some ocaml, I won't do it because I don't have the time right now
- we have a student that just started working on replacement of paparazzi center written in Python (and with a different way to save sessions)

So in the meantime, I suggest to save your session before leaving the paparazzi center or choose "keep current". Sorry.

kevindehecker, 2016-02-23:
Oooh, ah new Python Center. I'll happily wait for that :)

dewagter, 2016-02-23:
I would suggest to add a line to say:
If you recently checked out a different branch, this will have changed your
conf. To keep this new conf and forget the previous conf from the other
branch press ...

kevindehecker, 2016-02-24:
If we are to keep the msgbox for now, I'm thinking more along the line of:

"Configuration changes to conf/conf.xml were either not saved during the last session, or external changes were applied (e.g. from a git branch).\n In the first case, please choose [Keep current] to save the new settings permanently. In the second case, please choose [Discard local changes] to reset to the new settings of that branch."

[Keep current] [Discard local changes] [View changes]

Also, I'd change the default button to the safer [Keep current] option.

I can create a PR if this is ok

gautierhattenberger, 2016-02-24:
The first line is fine, but in the first case for instance, I prefer sometimes to discard local changes to go back to the previous version. The rest of the text implies that can't or shouldn't do that even if it is perfectly fine. So I suggest to at least add "or choose [Discard local changes] to restore previous settings.
Since my conf is not under revision control, the second case never occurs to me, so I don't know if it is relevant to keep current conf in this case.

kevindehecker, 2016-02-24:
I really think that is counter productive. It is not helpful to explain each and all possible use case in the text, simply because there are too many. (the second case also has all kinds of exceptions)
I think the new button caption should make it clear enough what to press in case of other use cases than the default recommend. I changed "please" to recommend, to reflect this:

"Configuration changes to conf/conf.xml were either not saved during the last session, or external changes were applied (e.g. from git).\n In the first case, we recommend to choose [Keep current] to save the new settings permanently. In the second case, we recommend to choose [Discard local changes] to reset to the new settings of that branch."

dewagter, 2016-02-25:
another proposition:

"Your conf/conf.xml is different than the last automatic backup. (e.g.
forgot to save or changed git branch).\n [Load conf.xml from current
branch]    [Load auto-backup conf from previous session]"

-Christophe

flixr, 2016-02-25:
@dewagter :+1: 
with maybe a minor shortening:
"Your conf/conf.xml is different than the last automatic backup.\n(e.g. forgot to save or changed git branch).\n [Use current conf.xml] [Use auto-backup from previous session]"

gautierhattenberger, 2016-02-25:
@dewagter @flixr I guess a shorter text is not a bad idea

dewagter, 2016-02-25:
I still do not like the work "current" a lot: that was the most confusing
word ("keep current") in the first place if you do not yet know how
paparazzi center works.
And 1 of the 3 reasons to get this dialog was still missing.

"Your conf/conf.xml is different than the last automatic backup.\n(e.g.
forgot to save, *manually edited* conf.xml or changed git branch).\n [Use
conf.xml (as is)] [Use auto-backup from previous session and overwrite
conf.xml]"

-Christophe

flixr, 2016-02-25:
good points, but in any case I would like to keep the button text shorter...

kevindehecker, 2016-02-25:
I have a big problem how "auto backup" is/was used. Normally (e.g. Microsoft Word), restoring the auto backup means restoring your last made changes (after a crash). In our case, it does the exact opposite. <insert obligate GUI design pattern comment>.

"Your conf/conf.xml is different than the last automatic backup.\n(e.g. forgot to save, manually edited conf.xml, or changed git branch).\n 

[Use conf.xml (as is)] [Discard local changes]"

dewagter, 2016-02-26:
can you explain a flow of actions where loading auto-backup does not load
the list of aircraft you last had in paparazzi center?
for me the only difference with how MS Word auto-backup works is that in
word when you open an auto-backup, it does not yet delete (overwrite) your
actual file.

-Christophe

dewagter, 2016-02-26:
Maybe with symlinks we could keep both:

 conf.xml -> conf.non-autobackup.xml
conf.autobackup.conf -> conf.xml

-Christophe

kevindehecker, 2016-02-26:
In MS Word, when it (automatically) restores the latest auto backup, it opens the copy of the original file that contains the last known/retained changes made by the user. It also presents you a list of older available revisions, in which one of the options is the last explicitly saved file, if for whatever reason you want to discard all the unsaved changes. Of course, MS Word does not have a notion of valid external changes through git or whatever... and that is what complicate things here.

In any case, the auto backup conf *should* contain changes made by the user. How to call the other kinds of conf backups I don't know. (maybe something like default / original / factory / ... conf).

dewagter, 2016-02-26:
auto-backup = last changes made by the user via *paparazzi center*
conf = last changes made by the user *directly to the conf file *(via
either notepad, or paparazzi center-save or via git)

paparazzi center does not know which is better as it depends on many many
possibilities.

the "issues":

 1) apparently the choice is not clear -> *suggested rephrases*
 2) unlike word it does not "open" the list of options to let you
select -> *suggest
to add a third button to open a diff tool between both*
 3) when you choose: you loose the other option forever: no way back:
meaning: you will loose data when you click the wrong button (unless you
nicely committed all your changes before starting paparazzi center) -> *suggest
to backup/symlink*

-Christophe

flixr, 2016-02-27:
there is already a "diff" button....

flixr, 2016-04-27:
Anyone not happy with the behavior or wording of the dialogs now?
https://github.com/paparazzi/paparazzi/pull/1591
https://github.com/paparazzi/paparazzi/pull/1640
Otherwise we can close this as done...

closed 2016-05-04 by flixr

podhrmic, 2016-05-05:
FYI I just tested the new dialog, and after reading through it actually makes sense and is easy to understand, so good job guys * thumbs up *