paparazzi/paparazzi #1555: [enhancement] Make GCS radio link indicator configurable

podhrmic, 2016-03-07:
Make the max `rc_rate` configurable either from airframe config file or from GCS config file. Here is the hardcoded value: https://github.com/paparazzi/paparazzi/blob/master/sw/ground_segment/cockpit/live.ml#L866

[image]

The reason is that some systems (like PPM radio) have frame rate typically up to 50Hz, but SBUS can go as far as 70Hz (so we want to know if we are lower than that), and in some cases as low as 20Hz (which is in our case a totally acceptable value). 

We want the GCS operator to be correctly informed, so it would be great to have the `max_rc_rate` configurable.

OpenUAS, 2016-03-22:
@podhrmic  :+1:  My OCAML fu a bit lacking though to fix it. This is a good chance maybe for you to dive in a bit as a small confined issue to solve.

OpenUAS, 2017-09-08:
Just bought a sBus out RX for on a LisaM just to test and fix the issue. Detail on how (not GCS side..) will follow.

OpenUAS, 2017-09-26:
RX arrived will mod it now for nomal non inverted and test on Lisa/M then test one of the solution come up with.

rijesha, 2017-10-19:
Well here is how to fix this on the GCS. https://github.com/rijesha/paparazzi/tree/rc_max_rate

It seems that the FBW_status message is sending the rate as a single unsigned int. a value of 5 corresponds to 50Hz. (so there is not a lot of precision) I have not tested this code.

gautierhattenberger, 2017-10-20:
The RC status bar has only five steps at the moment, that's why an integer with low resolution was enough. With old RC box using PPM signals, you could clearly see the quality level decreasing with distance. But with modern RC, you usually either have a good reception at full rate or nothing at all.
Still, it is not a big deal to adjust the rage or even have a higher resolution, but I have some doubts about the real benefit.
Can you make a pull request if you have something, it is easier to review for me. Thanks

rijesha, 2018-03-12:
@gautierhattenberger  can this be closed as well?

closed 2018-04-12 by gautierhattenberger