paparazzi/paparazzi #1556: Update JSBSim package to use JSBSim-1.0-RC2

podhrmic, 2016-03-09:
Paparazzi is currently using an older (and customized) version of JSBSim. Update it to the latest stable version ([1.0-rc2](https://sourceforge.net/projects/jsbsim/files/JSBSim_Source/JSBSim%20v1.0%20Release%20Candidate%202/)) to get bug fixes, new features and a standardized version of JSBSim (AFAIK we don't need any of the custom features that might be in the current package).

The standardized version will make further development of the simulator/hitl easier.

flixr, 2016-03-09:
We are actually using a more recent version of JSBSim, 1.0-rc2 is from 2009! And they have never made an actual release after that rc.
Currently the [paparazzi-jsbsim](https://github.com/paparazzi/paparazzi-portability-support/blob/master/debian/paparazzi-jsbsim/) package uses the version from 23 Feb 2015 and the only "customizations" are the addition of a pkg-config file and some fixes for Debian 8 (which were reported to the JSBSim team, so probably have already been incorporated since then).

podhrmic, 2016-03-09:
Ah, I didn't know that. So I guess then the question is - do we want to update to the newest version available at CVS ([February 2016](https://sourceforge.net/p/jsbsim/mailman/jsbsim-cvslogs/?viewmonth=201602)), or just keep the things as they are?

flixr, 2016-03-09:
Well, do you know if there are any new useful features or relevant fixes?

podhrmic, 2016-03-09:
Judging from the logs there are definitely *some* changes (I noticed a bunch of python scripts for testing and some more models), but I will look into it more.

podhrmic, 2016-03-11:
Changes in the latest version of JSBSIM (Feb 2016) compared to pprz version (23 Feb 2015):

In the newest version:
- minor changes in the example airplane xml files (doesn't really concern us as we don't use it)
- changed names of some variables (shorter names)
- cleaner code (i.e. more reusable code, proper function calls, generally looks like a little bit nicer code)
- added python test scripts (in case you are developing JSBSim you can run test cases on it)
- existing python scripts refactored (nicer python code)

I haven't noticed any added functionality in terms of simulation capability, the changes are mostly cleaner (and more robust?) code. In case we upgrade, I think we will have to update the JSBSim interface calls (in NPS), but that is only one file.

I am leaning towards the upgrade, since if it doesn't break anything it would be nicer to have the newer code (which also makes it easier to potentially contribute back to JSBSim), but I don't have strong opinion about it.

What do you think?

flixr, 2016-03-12:
If we really need to change how some JSBSim functions have to be called, it's a bit of problem:
An updated JSBSim package would then mean that NPS with JSBSim backend suddenly won't work/compile any more for all previous paparazzi versions (including latest stable release).

podhrmic, 2016-03-12:
Good point, I haven't realized that. So in that case we should probably close this issue and keep using what we have, at least for now. I don't know how would we solve it if we wanted to upgrade in the future though... Maybe have this done with releasing next stable version?

gautierhattenberger, 2016-03-14:
I think it is wise to close this and only consider upgrading for significant changes to the simulation part. Last time I did it was really a pain to keep the API the same so it can work with all versions of paparazzi/jsbsim.

flixr, 2016-03-14:
I agree with @gautierhattenberger, if an update really requires changing the function calls it should provide significant benefit.
If it's possible to update with only some minor changes AND it can be kept compatible with older versions, we can do it without problems...
The other option we could of course consider in the future (if keeping backwards compatibility is too hard otherwise) is to integrate JSBSim as a submodule as well instead of relying on one global system version for all paparazzi versions....

podhrmic, 2016-03-14:
I like the idea of JSBSim as a module. However, I ll close it for now, since no immediate action is needed.

closed 2016-03-14 by podhrmic