paparazzi/paparazzi #1559: Guidance speed reference

knmcguire, 2016-03-09:
The use_speed_ref in the module guidance_h does not seem to respond to RC commands anymore. I see that a new define is put in guidance_h.c (line 452):

#if GUIDANCE_H_USE_REF
#if GUIDANCE_H_USE_SPEED_REF
  **if (bit_is_set(guidance_h.sp.mask, 4) && bit_is_set(guidance_h.sp.mask, 5)) {**
    gh_update_ref_from_speed_sp(guidance_h.sp.speed);
  } else
#endif
    gh_update_ref_from_pos_sp(guidance_h.sp.pos);
#endif

Could anybody explain why this line in necessary? Thanks in advance

K

flixr, 2016-03-09:
What paparazzi version are you using?
I think this has already been reported in https://github.com/paparazzi/paparazzi/issues/1549 and fixed with https://github.com/paparazzi/paparazzi/commit/1e29cb623fc00e1fc34f377d5ccd60900dcef676

knmcguire, 2016-03-09:
Aha yes my branched version is exactly from 9 days ago. Sorry, my bad. I'll try it out these days and if there is anything else then I'll report it. Thanks for notifying me so quickly

closed 2016-03-09 by knmcguire