paparazzi/paparazzi #1582: Speech error during GCS startup

kevindehecker, 2016-03-30:
Sometimes, I get this:

RUN '/home/houjebek/paparazzi/sw/ground_segment/cockpit/gcs  -speech '
Failed to connect to Speech Dispatcher:
Error: Can't connect to unix socket /run/user/1000/speech-dispatcher/speechd.sock: No such file or directory. Autospawn: Autospawn failed. Speech Dispatcher refused to start with error code, stating this as a reason: Speech Dispatcher already running.

Does not seem to be a real big problem though. GCS starts and  functions, and the next time I start the error does not occur.

EwoudSmeur, 2016-03-30:
I have the same thing

flixr, 2016-03-30:
Not sure if there is much we can do about it, all the GCS does is call `spd-say` as system command:
https://github.com/paparazzi/paparazzi/blob/master/sw/ground_segment/cockpit/speech.ml#L32

Seems that `spd-say` prints that error if a speech dispacher is not running yet, although it will also start it...

OpenUAS, 2016-03-30:
Some insight https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=734392

OpenUAS, 2016-03-31:
a solution could be to look for speech dispatche process is running when starting the GCS with the -speech option, if running contine, else , shudder, a counter an as soon as ia flag is encounter start the first speech command only after the flag is true..

OpenUAS, 2017-02-01:
There is a new upstream package that should contain the fix. To get rid of the message a "speech dispatcher build" from current latest source should get rid of the issue. Not tested myself. Since there is a fix in upstream and this issue is annoying but not really a PPRZ issue per se an the speech output works anyhow, will close. Feel free to reopen if someone is willing and known to repend time on it.

closed 2017-02-01 by OpenUAS