paparazzi/paparazzi #1594: Compiling with Link Time Optimization flag

gautierhattenberger, 2016-04-04:
It seems that some boards are short in ROM space to fit the complete autopilot (like the Naze32). The recent Link Time Optimization can reduce the code size (and make it run faster) with the price of a slightly longer compilation time, especially during the link stage.

For instance, a rough test with the quad_flip32 airframe file gives a .bin of 123k (depending of the telemetry and settings enabled) with default settings, and is reduced to 119k just by adding `-flto` flag to `CFLAGS` and `LDFLAGS`.

I think it should be possible to enable this flag when the hardware or the user request it. My question is: where do we put this flag ? In every makefile type, only for embedded stuff, factorized in a makefile like `arm-embeded-toolchain` ?

flixr, 2016-04-05:
You can should already be able to do something like
`make AIRCRAFT=quad_flip32 ap.compile USER_CFLAGS=-flto`
Maybe we should extend that to USER_LDFLAGS?

Or we create a `ENABLE_LTO` flag or so, which we could then also set by default for some boards?
I wouldn't put something like that in the toolchain file... so probably in Makefile.stm and Makefile.lpc21?
We could also refactor the makefiles even more to pull out common things/flags (warnings, lto flag, etc..)

podhrmic, 2016-04-05:
Since it is board dependent, shouldn't it be in the board makefile?
How much slower is the compilation? If it is just a few seconds maybe we can have it enabled by default.

flixr, 2016-04-06:
> Since it is board dependent, shouldn't it be in the board makefile?

It's not really "board dependent" as you could use it for any target, but enabling a flag in the board makefile is what I meant with "set by default for some boards"...

closed 2016-05-01 by flixr