paparazzi/paparazzi #1599: Rotorcraft Sensor Calibration Refinements: ACCEL, MAG, MAG_CURRENT

hooperfly, 2016-04-14:
It's a tedious process to calibrate sensors for rotorcraft and it seems we might be able to refine the process a bit based on the migration to `modules`. Case in point, the updated `mag_current` calibration that uses a module vs. the `subsystem` approach is a nice refinement. After a bit of thought, it seems we might be able to do the same for the `accel

flixr, 2016-04-14:
Actually instead of running the calibration on a log, I would favour directly reading the messages from IVY.
I started a simple mag calibration script that does that in my calib_mag_live branch some time ago: https://github.com/flixr/paparazzi/commit/f67b7760e085e89ab7b0a53d84566350c90501c3

But I didn't continue since some ENAC students are apparently working on a nice calibration script/gui that does that. I had actually expected a pull request by now...

hooperfly, 2016-04-14:
Ok, that works too. Who would have status on the ENAC project?

gautierhattenberger, 2016-04-15:
The student project in itself is finished, but their code is not really usable right away, so we wanted to improve and polish it a bit before. The issue is we don't have much time for that. So at some point we may publish it as his and see who wants to work on it.
Remember they are first year students and it's their first programming project...

agressiva, 2016-04-16:
way not a module that make all calculations and save the values on the flash ? or a way to save the values on the settings tab using the persistent options ?

The way paparazzi is calibrated is very tedious and complicated for a ordinary user.

flixr, 2016-04-17:
Doing the optimization onboard would be a bit too heavy for most target MCUs right now... for a simple min/max calibration it would be doable.

hooperfly, 2016-04-19:
@flixr I looked at your calib_mag_live branch. I like it and was thinking about using Qt or other, instead of GTK. Any thoughts? Thinking about making a web based calibration tool using the PPRZLINK and the Flask stuff I've been using with the FRC.

flixr, 2016-04-19:
Unless there is significant benefit of using Qt for that case, I would suggest to stick with GTK.
We have no dependencies on Qt yet...

gautierhattenberger, 2016-04-19:
Our students are learning to use Qt only and they are using it in their projects as well. It is the case for the calibration tool they made, the pprz_center, and so on. I know it is not really helping, but at least now we can give them Python project and not Java only with the new Enac courses...

flixr, 2016-12-29:
See #1977

hooperfly, 2016-12-29:
@flixr I'll try out the new python mag tool referenced in #1977

OpenUAS, 2017-05-31:
@hooperfly Any update on this (issue)?

podhrmic, 2017-08-26:
Can we close this or do we need further updates/refinements? @hooperfly @flixr

hooperfly, 2017-08-29:
@podhrmic well we have added more options for calibration. I'll review the current process as I have a few aircraft to calibrate this week. I'll leave my thoughts here when I'm done. Thanks for the ping.

gautierhattenberger, 2021-12-13:
We still need a proper HMI, but closing this proposol

closed 2021-12-13 by gautierhattenberger