paparazzi/paparazzi #1614: module target selection not working in some cases

flixr, 2016-04-20:
While target inversion seemed to be fixed with #1583, it still doesn't seem to work as desired/expected in all cases:
- changing `target="ap"` to `target="!sim|nps"` is still supposed to add it to the `ap` target, but that is not the case for e.g. https://github.com/paparazzi/paparazzi/commit/3afcce0737aaf1acbcc4a00503f8382358c88d46
- when a module is in one firmware section (e.g. telelemetry_transparent in rotorcraft firmware), it is also added to all allowed targets in a different firmware (e.g. test_progs). Expected would be that the module is only added for the targets in the firmware section it is part of.

flixr, 2016-04-21:
Btw, this is why some targets like `demo_ahrs_actuators` are failing...

gautierhattenberger, 2016-04-26:
It is actually pretty tricky when a module has several makefile sections with different targets (like telemetry_transparent).
When building the list of possible targets, it is loosing the initial logical expression:
target="!sim|nps" target="ap" should be "not(sim||nps) || ap"
but it is seen as the list "!sim, nps, ap" which is understood as "not(sim||nps||ap)"
So I need to find an other way to store and test the list.

flixr, 2016-04-29:
Should be fixed with #1643

closed 2016-04-29 by flixr