paparazzi/paparazzi #1616: InsideSector function not generated properly

podhrmic, 2016-04-20:
Hi, 
as mentioned in the [wiki](http://wiki.paparazziuav.org/wiki/Flight_Plans#Sectors) the static sectors have to be defined **clockwise** and be **convex** which is actually really limiting from the user standpoint (I just ran into that issue).

The dynamic sectors seem to work fine no matter on the order their corners are defined, and no matter whether they are convex or not (since the used [algorithm](http://alienryderflex.com/polygon/) is really general).

Would be possible to fix the static sector function generator to work the same way as for dynamic sectors (i.e. to be able to deal with non-clockwise and non-convex polygons)?

gautierhattenberger, 2016-04-21:
The issue is named properly I think. The static algorithm is more efficient but also more limited (need to be convex on the north-sud axis, cannot be modified at runtime).
The dynamic version is potentially slower, but use the actual waypoint position at runtime.

If you want a static sector that works like the dynamic one, why not using a dynamic sector with hidden waypoints (prefixed by `_`) ?
Their is no real need to make the static version less efficient I think.

flixr, 2016-04-21:
> The issue is named properly I think.

I think you mean NOT properly named...

gautierhattenberger, 2016-04-21:
yes indeed...

podhrmic, 2016-04-21:
@gautierhattenberger thanks for the idea, I haven't realized that. That should be sufficient for our application (at least for now). Just FYI I am adding a "geofencing" feature - essentially a sector and an exception checking whether you are inside the sector, the difference is that it is required and can't be omitted from the flight plan. Hence being more flexible with defining static sectors would be helpful.

Anyway, would you be interested in replacing/enhancing the distance_to_home with the "compulsory" flight area sector?

gautierhattenberger, 2016-04-21:
I think it can be useful when the system is meant to be used by end-users as a commercial product, but for developers or testers, I personally don't want to think too much about this before flying, the protection circle is good enough. So if you have something, maybe you can share it to discuss the implementation, but for now I would prefer an optional solution.

podhrmic, 2016-04-21:
OK, sounds good - I will push what I have hopefully in next few days.

gautierhattenberger, 2016-10-12:
I suggest to close this. Geofincing and non-convex sectors have already been improved.

podhrmic, 2016-10-12:
I agree. What do you mean by non-convex sectors improvements? Last time I checked only convex sectors worked for geofencing, but it turned out to be satisfactory for all known use-cases.

gautierhattenberger, 2016-10-12:
"dynamic" sectors are supporting non-convex polygons. I guess they can also be used for geofencing.

closed 2016-10-12 by gautierhattenberger