paparazzi/paparazzi #1617: Bebop 1 GPS not working in master

dewagter, 2016-04-20:
- GPS port OK
 - GPS is configured
 - NMEA is decoded (incl PERCRV msg)

Abi message not sent or used: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/gps/gps_nmea.c#L108

any tips?

_____

ABI_NOT_SENT (in the else)!
parse 65 GSV
GSV: "GPGSV,4,3,13,30,12,319,19,15,11,037,,26,11,174,21,13,03,005,,1*63" 
p_GSV() - 4 sentences
p_GSV() - sentence=3
p_GSV() - num_sat=13
p_GSV() - GPS 8 PRN=30 elev=12 azim=319 snr=19
p_GSV() - GPS 9 PRN=15 elev=11 azim=37 snr=0
p_GSV() - GPS 10 PRN=26 elev=11 azim=174 snr=21
p_GSV() - GPS 11 PRN=13 elev=3 azim=5 snr=0

dewagter, 2016-04-20:
when decoding, all is filled in the gps_nmea, but the ABI function sends gps. Who should fill gps?

kirkscheper, 2016-04-20:
I guess it should be sending GPS_nmea

Kirk

dewagter, 2016-04-20:
gps_nmea is the wrong struct type for the ABI message. So I guess it must be copied?

kirkscheper, 2016-04-20:
Check this:
https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/gps/gps_datalink.c

kirkscheper, 2016-04-20:
Or send &GPS_nmea.state

Kirk

closed 2016-04-20 by flixr

dewagter, 2016-04-20:
> Closed #1617 <https://github.com/paparazzi/paparazzi/issues/1617> via
> 10fb395

@felix @kirk: Thanks

-Christophe

commit 10fb395a (2016-04-20): [gps] fix sending of ABI GPS messages in nmea, sirf and skytraq

should fix #1617