paparazzi/paparazzi #1627: vertical climb mode keeps climbing

EwoudSmeur, 2016-04-22:
If adaptive throttle is disabled, rotorcraft are very sensitive to the nominal hover throttle in nav climb mode. If the nominal hover throttle is too high, a nav_climb_speed of -1 can result in the drone ascending to infinity.

This is observed in reality as well as in sim. The problem seems to be that there is only a P gain active in addition to the nominal hover throttle...

flixr, 2016-04-22:
The difference between using adaptive throttle and fixed nominal hover throttle is only in the feedforward part.
Feedback loop is always the same PID loop: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_v.c#L452

dewagter, 2016-04-22:
In climb mode only the D gain is effective + nominal throttle. A large
error in nominal throttle requires a large error in climb rate to
compensate. That is why all those quads take off so slowly at 1m/s command.

flixr, 2016-04-22:
While the vertical climb control is not ideal, the P and I gains should still be effective as the z position setpoint (`guidance_v_z_ref`) is also advanced according to the speed setpoint in `gv_update_ref_from_zd_sp`.

dewagter, 2016-04-23:
they are not because https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_v.c#L296 constantly resets the reference model to the current altitude, so while P and I are active, they steer on a zero error and are not effective.

flixr, 2016-04-23:
Yes they are, as it doesn't set the z_ref to the current altitude (which is just passed to limit the max distance the z_ref is advanced), see https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_v_ref.c#L132

flixr, 2016-05-18:
@EwoudSmeur @dewagter any more insights here?

dewagter, 2016-05-19:
If the P and I gains are active while in climbing mode, then I think this
issue can be closed, and it should '''just''' be a matter of tuning.

___________________________________________

I should still make some calculations. e.g. the normal bebops with the 1
m/s climb rate command do not take-off properly (even with adaptive
throttle): they keep bouncing in ground effect for at least 10 seconds
before starting to very slowly climb. With a 2.5m/s climb command they
instantly start climbing. This is very likely due to an error in
`nominal_throttle` being too low. A higher p/i-gain on altitude should
solve this. But the current p-gain is ideal for normal flight though. So
for the take-off this is now solved (I should say '''hacked''') by
commanding an artificially high climb rate which is never reached (at least
not in the flightplans with a deroute when the altitude is sufficient).

Since the altitude error is limited, whenever the `GUIDANCE_V_MAX_ERR_Z *
P-gain + GUIDANCE_V_MAX_SUM_ERR * I-gain` is smaller than the error in
`nominal_throttle`, it keeps climbing/descending infinitely.

___________________________________________

-Christophe

flixr, 2016-05-19:
It work just like moving the vertical position setpoint at the given speed, if you want a "real" velocity controller it is a different matter...

dewagter, 2016-05-19:
BTW: in general I highly prefer integrating the Error * I-gain and limiting
this value (which is the trim setting and can be understood) than to
integrate the err and limiting this meaningless value to another
meaningless bound and multiplying with the gain afterwards.

_________________

Here are my calculations to what happens with bebop and outback
_________________

I-gain: 13 (bebop)

1 second x 1 meter too low: 512 * (8-bit-frac) = 130 000
130 000 x 13 / 16bit = 25 / PPRZ_MAX = 0.2% throttle.

5 sec 3 meter low = 2 000 000 (GUIDANCE_V_MAX_SUM_ERR)
-> 390/9600 = 4% extra throttle

P-gain: 283

Max Err = 10m * (8bit-frac) = 2560
* gain / 7bit -> +/- 58% throttle

1 second after take-off with too low norminal throttle (staying on the
ground): 6% extra throttle ( = still on the ground), ...

_________________

outback : I-gain = 6 -> max throttle increment = 1.9%
P-gain = 75 -> max throttle increment = 15%

=> nominal throttle error of 17% yields infinite climbing, and error of
more than 2% yields a steady state altitude error.

-Christophe

flixr, 2016-05-19:
> BTW: in general I highly prefer integrating the Error * I-gain and limiting this value (which is the trim setting and can be understood) than to integrate the err and limiting this meaningless value to another meaningless bound and multiplying with the gain afterwards.

Yes, that makes more sense in general... feel free to make a pull request for this.

But coming back the original issue: if you disable adaptive throttle (why?) and nominal throttle is set too high, then you get a too high throttle due the feedforward term. This would need to be compensated by the feedback terms.

flixr, 2016-05-19:
Also have a look at the vertical reference model and it's parameters: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_v_ref.c#L50
Default max velocity is 3 m/s, max acceleration 0.8g, maximum distance altitude setpoint is advanced in climb mode is 2m, etc...

dewagter, 2016-05-19:
Thank you for the tip: if the reference model is limited is Z_err=2, then
that finally explains why it can happen that during climb something keeps
climbing while during hover the gains are sufficient as the P-gain has 5x
more authority (max-err * gain) in hover than during climb mode.

-Christophe

flixr, 2016-05-19:
I guess we should set the default MAX_Z_DIFF to 10m as well...

flixr, 2016-06-14:
any new updates/insights here?

dewagter, 2016-06-15:
The behavior is now understood, so for me this can be closed.
A small improvement could be to set the same limit for the control and the
reference model, which would cause climb to behave more like the hover.

-Christophe

flixr, 2016-06-15:
currently we have the defaults:
- `GUIDANCE_V_REF_MAX_Z_DIFF` at 2m: maximum distance altitude setpoint is advanced in climb mode
- `GUIDANCE_V_MAX_ERR_Z` at 10m

So set both to 10m or some other value?

dewagter, 2016-06-15:
can you
#define GUIDANCE_V_REF_MAX_Z_DIFF GUIDANCE_V_MAX_ERR_Z
?

-Christophe

flixr, 2016-06-15:
no, we could only make `GUIDANCE_V_MAX_ERR_Z` and `GUIDANCE_V_MIN_ERR_Z` default to +-`GUIDANCE_V_REF_MAX_Z_DIFF`

commit 3d60dc65 (2016-06-16): [rotorcraft] GUIDANCE_V_REF_MAX_Z_DIFF defaults to 10.0m

just like GUIDANCE_V_MAX_ERR_Z

This should result in less unexpected behavior and closes #1627
closed 2016-06-16 by flixr