paparazzi/paparazzi #1652: [NPS] Fixedwing goes to HOME mode immediately after take-off

podhrmic, 2016-05-04:
Using latest master and default configuration, Ubuntu 12.04:
[image]

Choose Bixler with the default flightplan (versatile.xml) and target NPS, compile and Simulate. The plane takes off and then immediately goes to HOME mode, even though it is not supposed to.
[image]

Ideas?

flixr, 2016-05-04:
Seems to be something introduced with #1637, not sure exactly what yet though...

closed 2016-05-04 by flixr

podhrmic, 2016-05-04:
Nice! Thanks for the quick response:)

commit f5a37d0d (2016-05-04): [gps] fix utm_float_from_gps

this fixes utm zone in utm_float_from_gps, this was a regression from #1637

fixes #1652