paparazzi/paparazzi #1668: GCS arg of section programs not working

dewagter, 2016-05-16:
This does NOT work for me.

  <section name="programs">
    <program name="GCS" command="sw/ground_segment/cockpit/gcs">
      <arg flag="-layout" constant="bottom_settings.xml"/>
    </program>
  </section>

The following DOES work (except for REPLAY and SIMULATION)

  <section name="sessions">
    <session name="FOO">
      <program name="GCS">
        <arg flag="-speech" />
        <arg flag="-layout"  constant="bottom_settings.xml" />
      </program>
    </session>
  </section>

dewagter, 2016-05-16:
discussion started in #1600

flixr, 2016-05-16:
The first is used if you start the GCS from the tools menu, the second if you start the session FOO

dewagter, 2016-05-16:
Any trick to set the layout for the SIMULATION then?

-Christophe

flixr, 2016-05-16:
I think that is currently hardcoded, but we can probably change that to use the entry with arguments from the control_panel.xml

PR #1671 (2016-05-18): [pprz_center] also use GCS program args for Simulation session
and Replay session.
This allows to e.g. set the desired default options for the GCS (like layout, speech, etc..) in the programs section of the control_panel.xml file
and they will also be used for the Simulation and Replay sessions and not only when started from the tools menu.

should close #1668
commit 532ee0c3 (2016-05-18): [pprz_center] also use GCS program args for Simulation session (#1671)

and Replay session.
This allows to e.g. set the desired default options for the GCS (like layout, speech, etc..) in the programs section of the control_panel.xml file
and they will also be used for the Simulation and Replay sessions and not only when started from the tools menu.

closes #1668
should close #1668
closed 2016-05-18 by flixr