paparazzi/paparazzi #1674: [telemetry] "module" attribute of messages not considered for modules specified in flight plan can't

flixr, 2016-05-18:
When parsing the telemetry xml file, if a message has a "module" attribute, it is only added if the specified module is loaded.

This doesn't seem to be the case for modules specified in the flight plan, since `get_modules_name` in gen_common.ml only calls `get_modules_of_airframe` but not `get_modules_of_flight_plan`

gautierhattenberger, 2016-05-18:
Stupid question: this mechanism was made when telemetry scheduler was a macro. With the new system it doesn't fail when a message is not register. Also it seems that no telemetry files are using this anymore.
Shouldn't we just remove this ?

flixr, 2016-05-18:
Yep, not really needed any more, so we might as well remove it.

Just came across the `get_module_name` function and noticed this discrepancy....

closed 2016-05-21 by flixr