paparazzi/paparazzi #1681: OpenCV - bebop - __gnu_cxx::__concurrence_broadcast_error

TomasDuro, 2016-05-22:
I designed a module to use the frames from the bebop front camera. The module compiled well but when I ran it through the GCS it wouldn't work. Running it through the terminal gave me an automatic error message:
"__gnu_cxx::__concurrence_broadcast_error”
The only way to solve this problem I found was to remove the -static flag from the bebop.makefile (line:"$(TARGET).LDFLAGS += -static"). By doing so it allowed the module to work properly.

flixr, 2016-07-14:
@TomasDuro @kirkscheper @rmeertens is this still an issue or can we close it?

rmeertens, 2016-07-14:
@TomasDuro I vaguely remember you found something that made it work (or... found away around this problem). Care to share what it was?

flixr, 2016-07-14:
The other OpenCV things on bebop seem to compile fine... at least in current master (v5.9_devel-767-g10df83e)

rmeertens, 2016-07-14:
Problem was that it crashed on the drone...

flixr, 2016-07-14:
Right... I don't have that stuff to test, so I'll leave this to you....

OpenUAS, 2017-03-02:
Could not reproduce uded OpenCV v3.1 GCC v5.4 on Bebop 1 and Bebop 2

closed 2017-03-02 by OpenUAS