paparazzi/paparazzi #1698: set_nice_level in vision_thread

knmcguire, 2016-06-01:
Hi all,

Since the new vision_thread has been implemented, the functions called by it (cv_add) were running a lot slower before the implementation. I just figured out that the set_nice_level (line 170) in modules/computer_vision/vision_thread.c, is put on 10, which gives the vision thread a very low priority in the timing to reduce CPU usage I presume.

I definitely get the idea of controlling the nice level to protect the rest of the processes of the autopilot, for some vision algorithms this will do just fine, but for optical flow control, the loop rate should be as high as possible. A nice level of 10 drastically reduces the performance.

If I want to make a pull request about this, what is the best approach? To replace the fixed number in set_nice_level with a define (standard: 5, other can be specified in airframe file)?

Thanks in advance

dewagter, 2016-06-02:
Since this only applies to ardrone and bebop, and both have shown to fly well even with total vision processing overload, and that vision results are used in the control loop so vision becomes part of the (position) stabilization, I would suggest to (also) change the default value.
Obviously I do not mind if it can be changed (setting/airframe)

wilcoschoneveld, 2016-06-24:
I believe the pull request solved this issue.

closed 2016-06-25 by flixr