paparazzi/paparazzi #1703: Strings Deprecated Warning for Ocaml ≥ 4.02

rijeshaugustine, 2016-06-03:
As of Ocaml v4.02 Strings have been deprecated in favour of Bytes. 
https://ocaml.org/releases/4.02.html

This results in a build warning when compiling Paparazzi.

Changing Strings to Bytes is not sufficient to silence the warnings, as it would break compilation for everyone with OCaml < 4.02.0

@flixr was looing into using the camlp4 preprocessor to IFDEF around this issue... https://github.com/flixr/paparazzi/commit/007deabafe5fecc6ea7a1178566e69de4b804539

flixr, 2016-06-03:
We could also use `Bytes` everywhere and add `module Bytes = String` for OCaml < 4.02, but that would mean we would have to preprocess every file with camlp4...
So it seems like the nicer option would be to add something like a compat module with `bytes_create` etc. functions and only preprocess the compat.ml and compat.mli with camlp4...

rijeshaugustine, 2016-06-04:
Is this an ok way to fix the makefile?
https://github.com/rijeshaugustine/paparazzi/commit/af911cd1666cf4d0603f0e7368e5081e63409574

rijeshaugustine, 2016-06-05:
Ok so I think I fixed the Makefile. I used a shell if statement to only compile files listed in PP_SRC with the preprocessor.  https://github.com/rijeshaugustine/paparazzi/commit/81f9dfb182400af08808f1c31deaf6eb27bb23d4

I started to modify the rest of the .ml files.

PPRZLINK will need to be modified. It compiles first so compat.ml will have to be copied into that repo as well and the Makefile will need to be set up for the ocampl4. Unless there is a better way of doing it?

flixr, 2016-06-06:
Thanks for the work @rijeshaugustine !
Can you plz make a pull request?
pprzlink will have to be updated separately...

rijeshaugustine, 2016-06-09:
For pprzlink just adding the compat module to it will cause another warning when building sw/lib/ocaml due to compat.cmo already existing. Can this be rectified by telling the sw/lib/ocaml/Makefile to ignore compat.ml if compat.cmo exits in sw/ext/pprzlink/*/ocaml/ or will that cause other issues in the future?

podhrmic, 2016-06-09:
Just an idea - can we update the ocaml version that is installed with paparazzi (i.e. package update)? That way we wouldn't have to worry about the backward compatibility (<4.02) I suppose?  (I am assuming that even older OS like Ubuntu 12.04 can upgrade to the >4.02 Ocaml)

flixr, 2016-06-09:
@podhrmic that would be only possible if you manually install a newer OCaml version (from source or via opam)...
Debian (Ubuntu) packages are available in these versions: http://packages.ubuntu.com/search?keywords=ocaml
So 4.02 is only available since latest xenial

gautierhattenberger, 2016-06-09:
What I did for the `Debug` module is that I just made a specific one for pprzlink with a different name...

rijeshaugustine, 2016-06-11:
made a pull request for pprzlink https://github.com/paparazzi/pprzlink/pull/23

flixr, 2016-06-14:
Thanks! closing this as the relevant pull requests have been merged...

closed 2016-06-14 by flixr