paparazzi/paparazzi #1710: Waypoint LLH not updated after geo-init

dewagter, 2016-06-10:
struct LlaCoor_i *lla = waypoint_get_lla

still provides the initial flightplan origin even after a GEO-Init

flixr, 2016-06-10:
I guess you set the waypoint as a global one? Then it doesn't change with geo-init...
See http://docs.paparazziuav.org/latest/subsystems_2navigation_2waypoints_8h.html#a2ed9112b04893275873729d03176f490

flixr, 2016-06-10:
See also https://github.com/paparazzi/paparazzi/pull/1065

dewagter, 2016-06-13:
When LLH are defined: I know waypoints do not move: perfect
When XY are defined: they update nicely in the paparazzi ground station: perfect

But:

After a geo-init: the ```waypoint_get_lla``` function (e.g. called from mavlink waypoint.c) still returns the flightplan origin+xy.

flixr, 2016-06-13:
I don't understand what you mean: if the waypoint has it's global LLA coordinates set, then it will always return that (and that equals flight plan origin + xy).

Maybe you can provide a concrete reproducible example of where it goes wrong?

commit 5b8f93eb (2016-06-13): [waypoints] fix waypoint_get_lla

should fix #1710
closed 2016-06-13 by flixr