paparazzi/paparazzi #1721: [airframe generator] throw an error when a requested module doesn't exsit

kirkscheper, 2016-06-14:
Title says it all

gautierhattenberger, 2016-06-14:
Obviously, if you ask for something that doesn't exist, it will complain. Which kind of behavior would you expect when loading an invalid module/subsystem ?

flixr, 2016-06-14:
it seems that it currently does not complain anymore, just silents skips the invalid one...

gautierhattenberger, 2016-06-14:
really ? If try to load a module like `bla` it just fails. Actually it fails because the module doesn't exist and it falls back to subsystem, which doesn't exist either, so make is failing.

kirkscheper, 2016-06-14:
What I did today was <module name="cv_optickflow"\> which was a typo. It gave me no warning at all. just built and uploaded.

gautierhattenberger, 2016-06-14:
can you provide the example airframe and paparazzi version you are using ?

kirkscheper, 2016-06-14:
paparazzi version was a copy of master from today with the bebop 
airframe from the examples folder with the added line I mentioned.

gautierhattenberger, 2016-06-14:
hum, actually, if you make a typo in a module from the firmware section, but not from the modules section. I'll look into this.

flixr, 2016-06-14:
It doesn't fail if the bad module is in the `<module/>` section

commit 63c6884a (2016-06-15): [buid] fail if module doesn't exist

close #1721
closed 2016-06-15 by gautierhattenberger