paparazzi/paparazzi #1747: [gcs] map tiles mixed up

bartslinger, 2016-06-19:
From the menu in the GCS, I selected Maps->Dump map of Tiles. The generated *.png already shows the tiles mixed up, and that's also how they appear in the GCS if I load the file. Tried both google and bing as map source, but that did not solve the problem.
[image]

OpenUAS, 2017-02-14:
Will test and see to reproduce asap

gautierhattenberger, 2017-07-25:
did anyone managed to reproduce this bug ?

podhrmic, 2017-07-25:
Yes, this is what it looks like on Ubuntu 16.04 and master pprz.

First, this is the GCS view with a rectangle select for the tile dump:
[image]

This is the resulting tile dump file:
[image]

Obviously wrong. On the other side, I didn't know about that feature (Dump Tiles) until I read this issue, so perhaps it can be removed if it is too hard to fix?

OpenUAS, 2017-08-03:
I yes, could reproduce, on latest, de feature is uest and looking at it maybe has to do with wrong tile zoomlevel, recal that changed to higher in the past, but maybe not this part.

gautierhattenberger, 2018-06-06:
closing as won't be fixed in current GCS
see https://github.com/paparazzi/paparazzi/projects/2

closed 2018-06-06 by gautierhattenberger