paparazzi/paparazzi #1762: Bebop bottom camera not functioning properly

wilcoschoneveld, 2016-06-22:
Today we updated the firmware of two bebop drones (one from version 1.98, the other 2.0) to the latest 3.2.0. After updating, the bottom camera does not work as expected. We tried to isolate the issue by testing on both drones and performing a factory reset (by holding the power button for 10 sec).

The problems encountered are as follows;
- The bottom camera cannot initialize (common)
- The bottom camera runs but shows only 50% FOV (sometimes, after factory reset)
- Running ap.elf via telnet shows the following error; `Select timeout on /dev/video0`

Additional information:
- We are using the tudelft_course2016_bebop_avoider airframe, but using only the video_thread, cv_colorfilter and video_rtp_stream modules.
- The front camera works as expected, even when trying to run both camera's in parallel
- Even when the bottom camera is working somewhat (with half of the expected field of view) the video stream delay is longer than before)

dewagter, 2016-06-22:
Probably the resolution changed. If you try to open at the wrong resolution
then you can open the video device

-Christophe

kirkscheper, 2016-06-26:
is there any progress with this issue? It is blocking for a project we are working on.

wilcoschoneveld, 2016-06-26:
I'm going to quickly look at the resolution setting tomorrow, i'll report back.

fvantienen, 2016-06-26:
As I explained in my other PR with my new branch you have more settings and everything works fine.

kirkscheper, 2016-06-27:
> everything works fine.

@fvantienen the front camera doesn't work on your PR as @wilcoschoneveld stated in that thread, has that changed since last week? If not do you have an estimate as to when will it be working?

closed 2016-06-29 by wilcoschoneveld