paparazzi/paparazzi #1770: Bebop: detect firmware version at runtime to choose correct motor indexing

flixr, 2016-06-27:
To avoid errors of wrong motor ordering in the airframe files depending on Bebop firmware version, there should be "hardcoded" assignments and the correct one chosen at runtime.

See also #1745

OpenUAS, 2016-08-03:
+1 , but no time ATM sorry

OpenUAS, 2017-03-03:
While I fully agree that indeed is the nicest solution, nobody as it seems has time to do that. All 30+ TU Bebop1 airframes have firmware 3.3 now BTW. For others in the world. we could make an BB1v3.2 example airframe with the less optimal fix. Since there is a script now to forever prevent the annoying 4x button press on FW 3.3+ the urge to stay at v3.2 was very low. It is unlikely there will be a new change to firmware anyhow.

closed 2017-03-03 by OpenUAS