paparazzi/paparazzi #1772: [Feature Request] Run opticflow on multiple cameras symulteaneously

kirkscheper, 2016-06-27:
With the current implementation only one opticflow module can run at a time on one camera. I am not sure how popular something like this would be but I can imagine that it may be interesting to be able to run opticflow on multiple cameras (eg bottom camera and forward looking camera) simultaneously.

flixr, 2016-06-27:
You can't really run multiple instances of the same module.
Instead one would have to do something like:
- create an array of the opticflow struct to hold the parameters for each cam
- somehow deal with the sets of configuration options (defines) for each one
- extend the module to register a callback for each camera (and probably create multiple threads for processing)

matteobarbera, 2020-11-09:
I'm trying to tackle this issue, while trying to keep duplicate code to a minimum. My current idea, however, requires changes to `cv.h` and `cv.c`:

- Add `uint8_t id` parameter to the `video_listener` struct
- Add `uint8_t id` as function parameters for the `cv_add_to_device`* functions
- Pass `listener->id` to the cv linked function in `cv_async_thread` and `cv_run_device` https://github.com/paparazzi/paparazzi/blob/5887bf9746cb52ca1d46631b8d7435dc7ce32ffb/sw/airborne/modules/computer_vision/cv.c#L166 https://github.com/paparazzi/paparazzi/blob/5887bf9746cb52ca1d46631b8d7435dc7ce32ffb/sw/airborne/modules/computer_vision/cv.c#L201

This way any computer vision function could take advantage of the id and be able to use it as an index for e.g. an array of opticflow structs. All cv files that need to deal with multiple cameras currently have to create extra functions to differentiate between the cameras, this solution could reduce unnecessary code and make many of these files easier to maintain (this isn't currently a huge issue).

Lots of files would have to be changed which might cause problems for a lot of people, but I think this sort of change is better than trying to work around the `cv.c` limitations and having to create lots of extra unnecessary code.

I cannot think of a way to avoid writing a lot of duplicate code to configure multiple cameras, as this is done from XML defines (and there are a lot of them). This limits the max number of cameras the opticflow module can use to 2 rather than an arbitrary number of cameras, which would be ideal.