paparazzi/paparazzi #1778: Video thread save shot produces bad images

mwmkuijpers, 2016-07-04:
When taking a shot periodically, the saved images seem to be corrupted. When looking at the video stream, the images are fine. How can I solve this?

flixr, 2016-07-04:
Could you please provide some more details?
- What do you mean by corrupted? Look weird or the actual file is corrupted and can't be displayed?
- What exact Paparazzi version (output of `./paparazzi_version`) are you using?
- What airframe/drone are you using? Which camera?

\cc @rmeertens

mwmkuijpers, 2016-07-04:
[image]

They look like the example
I am using v5.8.2_stable-37-g37bb9ee-dirty or ardrone2

flixr, 2016-07-04:
@rmeertens @fvantienen any ideas here?
@mwmkuijpers maybe you can try with latest master where this is all refactored?

rmeertens, 2016-07-04:
Assuming you are Marly (helped her a few days ago) I have some questions: 
- How old is this version? (What date? As @flixr  mentions, is it before or after the refactor?)
- Last time I looked at the code you tried saving shots of two cameras at once... if this is still the case it would be important to mention
- I recommended you to use the video_usb_logger and adjust that code. Did this work for you?

mwmkuijpers, 2016-07-05:
-I had some problems getting the new paparazzi version to work so i reinstalled everything, and now i am using the newest version
- I still want to use the 2 cameras at once, but even when i try it with just 1 camera the images look this way
- with the video_usb_logger the images for 1 camera look similar to the ones i made with the video_thread

dewagter, 2016-07-05:
One possible reason for corrupt jpeg is if a jpeg compression variable is
shared over the 2 threads.

mwmkuijpers, 2016-07-05:
For the video_usb_logger, I changed nothing, just added what I thought were the necessary lines of code in ardrone2.xml to get it working. So I am just using 1 camera for testing. This should give me proper images, right? (Assuming that this code was tested before and it worked properly)
So what else could it be?

rmeertens, 2016-07-05:
Mmm, strange. The video_usb_logger is something I made and that worked perfectly the last time I tested it :s
What did you think were the necessary lines of code to get it working?

mwmkuijpers, 2016-07-05:
<module name="logger_file"/>
    
    <module name="video_thread">
    </module>
    
    <module name="video_rtp_stream">
        <define name="VIEWVIDEO_CAMERA" value="front_camera"/>
    </module>
	<module name="video_usb_logger">
    <define name="VIDEO_USB_LOGGER_CAMERA" value="front_camera"/>
    </module>
  </modules>

This is how I have it in ardrone2.xml.

rmeertens, 2016-07-05:
what happens if you don't use the video_rtp_stream? And can you test this setup on a bebop?

mwmkuijpers, 2016-07-05:
If I dont use video_rtp_stream I get the following error: DEPENDENCY Error: Module video_usb_logger requires video_rtp_stream

And I will go to the mavlab to ask if I can

mwmkuijpers, 2016-07-05:
I tested it on the bebop and it works fine on the bebop..

dewagter, 2016-07-15:
only 1 instance jpeg_encode can be called at the same time:
 - video_stream_rtp , video_shot_bottom, video_shot_front all use the jpeg encoder in different threads.

rmeertens, 2016-07-15:
Ok, solution would be to remove the globals and give each thread its on jpeg encoder.

dewagter, 2016-07-15:
solved in

commit d255a62d (2016-07-15): ardrone stream front camera: closes #1778
closed 2016-07-15 by dewagter