paparazzi/paparazzi #1782: Code safety and security

podhrmic, 2016-07-07:
Hi, 
out of curiosity I used [Flawfinder](http://www.dwheeler.com/flawfinder/) to check the paparazzi code for possible security weaknesses. Although the tool has its limits, and the weaknesses are mostly related to a possible buffer overflow on linux systems (it is harder to "hack" bare-metal or minimal-OS system), I found it kind of interesting.
In practice it can be summarized to _If possible use **snprintf** or **vsnprintf** instead of **sprintf**_

The full results (in case you don't want to install the program) are attached below. The main question is - is there interest into looking into these possible security issues in more depth and fixing them? 
They in vast majority concern linux systems (ArDrone, bebob). By no means those are some critical issues, it is mostly about making the whole system more secure.

Thoughts? @flixr @gautierhattenberger 

[results_flawfinder.zip](https://github.com/paparazzi/paparazzi/files/351436/results_flawfinder.zip)

flixr, 2016-07-07:
flawfinder looks useful, although for our use case more general static code analysis tools are probably better suited, e.g. https://scan.coverity.com/projects/paparazzi-paparazzi

Regarding sprintf, very true... I tried to point it out in every new PR that added it, but it seems that most people don't really care that much about that.

gautierhattenberger, 2016-07-07:
Last time I did a static analysis, it was almost only harmless errors (like no checks for something that will never happen in real life).
But it is still useful to do such things from time to time and several configurations in order to cover more code.

podhrmic, 2016-07-07:
@flixr can you add me to the _coverity_ project? I am curious to look at the warnings/errors it shows.

flixr, 2016-07-07:
Done. Unfortunately there is no setting to allow anyone to see the report, you can only set the summary to public...

podhrmic, 2016-07-07:
Thanks! Looks like the flight code is mostly without defects (except few hits from the math folder). I am not sure if it makes sense worrying about ground code too, but it is cool to see that you configured it.

flixr, 2016-07-07:
You have to be aware though that only things that are compiled in `make test` are checked...

podhrmic, 2016-07-07:
Ah, I see. I haven't looked exactly into what exactly gets checked - but I guess we can expand the tests later if needed to get full coverage.

flixr, 2016-07-07:
I also just changed it from only running `make` to `make test`, but I'm not sure it actually covers the ARM builds (and not only the x86 NPS target)...

podhrmic, 2016-07-13:
Looks like the now we have a pretty decent defect density and also minimum defects detected in the flight code: https://scan.coverity.com/projects/paparazzi-paparazzi?tab=overview
Lets keep it that way - ideally once I week we push the new commits from master to `coverity_scan` branch and see what happens.

The only "problematic" section (besides the ground segment) is `arch/linux` and `computer_vision` folders - since I don't have the capability to test those, is there interest/will from other developers to fix the issues there?

flixr, 2016-07-13:
Is the airborne code that is cross compiled for ARM now really also checked? Since after adding that no new "defects" were found...
Did you test run that locally and look at the report?

podhrmic, 2016-07-13:
I did and it gave me the same results (see https://scan.coverity.com/projects/podhrmic-paparazzi?tab=overview )
If we want to be super sure I would "artificially introduce" a defect in the flight code and see if the scan will find it.

flixr, 2016-07-13:
Or you run only `make AIRCRAFT=foo ap.compile` and see if it picks up things that it noticed before when compiling NPS...
From the docs (https://scan.coverity.com/download) it sounds like it would clearly show that either it didn't pick up anything (without cov_configure) or that it has something...

podhrmic, 2016-07-14:
So it doesn't seem to detect my "artificial defect" a65b61099664c89daa57151393a8f6a91dac6e1e

When I tried to run it locally with `make AIRCRAFT` etc... it compiled the code but it complained that no coverity files were generated (and I configured it for ARM gcc).

@flixr do you want to give it a shot yourself? Otherwise I would send a question to the coverity support because looks like right now it doesn't really cover non-NPS targets....

podhrmic, 2016-07-14:
@flixr this should fix it: https://github.com/paparazzi/paparazzi/commit/b40c9474716929fd25d2bdbb9be291c55e34e106 (waiting for results)

flixr, 2016-07-14:
If that works, we should probably also set up `arm-linux-gnueabi-gcc`, `arm-linux-gnueabi-g++`, `arm-linux-gnueabihf-g++`, etc.

podhrmic, 2016-07-14:
AFAIK coverity lets you set up only one type of compiler per scan, so we would either change it manually in travis config when needed or have forks of pprz with different compliers. I ll keep you updated (the new build hasn't finished yet)

podhrmic, 2016-07-14:
@flixr it still hasn't built yet even though I pushed the changes yesterday - is there a way how you can check if travis built correctly?

flixr, 2016-07-14:
See branch overview: https://travis-ci.org/paparazzi/paparazzi/branches

podhrmic, 2016-07-14:
Ha! 

```
tests/aircrafts/01_compile_all_aircrafts.t .. 122/125 # compiling AIRCRAFT: [yapa_chimu_spi] TARGET: [ap]

tests/aircrafts/01_compile_all_aircrafts.t .. 123/125 # compiling AIRCRAFT: [yapa_chimu_spi] TARGET: [sim]

The job exceeded the maximum time limit for jobs, and has been terminated.
```

Looks like I might have to submit a subset of `conf_test` for coverity....

podhrmic, 2016-07-14:
FYI I manually submitted a build - see what new defects it found: https://scan.coverity.com/projects/paparazzi-paparazzi?tab=overview

( now it doesn't recognize the NPS targets for change)

podhrmic, 2016-08-10:
So I am roughly weekly pushing new commits from master to the coverity_scan branch. Right now it has almost identical conf_test.xml, but I had to remove a few aircrafts because the builds were taking too long. 

@flixr is there a better way how to handle conflicts in travis.yml file? I do `git rebase master` and then I have to manually fix the travis file - not a big deal, it is just annoying. 

I am fixing the defects slowly (see the project stats https://scan.coverity.com/projects/paparazzi-paparazzi?tab=overview and the Coverity fix commits) - but more help with linux arch and ground code would appreciated. 

@gautierhattenberger there are some defects related to the filesystem and datalogging used by Apogee, would you have time to look at it?

flixr, 2016-08-10:
instead of rebasing you can just merge master into the coverity branch...

podhrmic, 2018-04-14:
I think we can close this, as we are using coverity scan occasionally. Flawfinder is nice, but I don't think there is currently manpower to change the codebase based on its findings. We could discuss security/safety in a separate issue.

closed 2018-04-14 by podhrmic