paparazzi/paparazzi #1783: MISRA-C compliance

podhrmic, 2016-07-07:
In order to make Paparazzi more appealing for semi-professional or professional users (i.e. people or companies using UAVs to collect data and/or provide some services), it would be useful to show that our code is reliable and safe. One to do so is to have the airborne code confirm to [MISRA-C](https://en.wikipedia.org/wiki/MISRA_C). If the airborne code is MISRA-C compliant, then it is (at least in theory) easier to obtain flight authorization from the local authority (for example FAA in US). 

I analysed the airborne code for MISRA-C compliance, and actually the results are not too bad. The issues are ranging from _The goto statement shouldn't be used_  and _Assembly language shall be encapsulated and isolated_ to _Identifiers in the same namespace with overlapping visibility should be typographically unambiguous_ etc. but mostly it is a syntactic issue (below is a snapshot of the analysis result):
[image]

Here is shown what files/folder contains the most violations of the standard:
[image]

As with any standard, updating the code to conform to the standard would be mostly tedious and repetitive work (fixing similar issues in multiple files), but certainly not too difficult. Also, it might make sense to not comply with all MISRA directives, but only with the subset that makes most sense.

Anyway, is there interest from other developers? @flixr @gautierhattenberger @deonblaauw

deonblaauw, 2016-07-07:
I would be interested to do this, yes. Good initiative @podhrmic ! If I'm not mistaken, you are using "Understand for C" to do the analysis there? That GUI seems familiar... MISRA-C compliance is a very good step in the right direction if you want to convince certification authorities.

flixr, 2016-07-07:
That would be nice. Also if you can distill some of the most important rules out, updates to our coding style guide would be appreciated.

gautierhattenberger, 2016-07-07:
Is there any open-source tools/method comparable to MISRA-C ? It seems to be more or less a closed standard.

deonblaauw, 2016-07-07:
I don't know, but you can easily find the guidelines: http://caxapa.ru/thumbs/468328/misra-c-2004.pdf

podhrmic, 2016-07-07:
@deonblaauw yes I used _Understand for C_ - I have a 14 day trial version. I don't know if they were willing to give out a "free" licence for Paparazzi? 

Btw in the graph - the darker color, the more deviations from the rules.

@flixr I will see what parts of the rules can be distilled into a style guide. 

@gautierhattenberger There is [this](http://cwe.mitre.org/index.html) but it is more related to security of the code. Otherwise I found [cppcheck](http://cppcheck.sourceforge.net/) which is free, but doesn't explicitly help with MISRA (and I didnt make it work yet). Btw they sell MISRA-C standard for [£15](http://www.misra.org.uk/Buyonline/tabid/58/Default.aspx)

Also, the guide below is useful for understanding how the standard works - again, I am hoping to make MISRA-C compliance more feasible in the future by adhering to the rules/requirements where reasonable (so mostly changing our style guide/naming conventions/coding style). The documentation part is another topic.
 
[MISRA-C3-0001.pdf](https://github.com/paparazzi/paparazzi/files/352654/MISRA-C3-0001.pdf)

flixr, 2016-07-07:
cppcheck is especially nice to use as an IDE plugin... I use that at work...

podhrmic, 2016-07-07:
Nice! Do you use Eclipse IDE?

flixr, 2016-07-07:
No, I use CLion at work (we use CMake for everything).

podhrmic, 2016-07-08:
@deonblaauw do you have a recommendation about which of the MISRA directives we should focus on first?

deonblaauw, 2016-07-10:
@podhrmic I would suggest to initially only focus on the required rules (if you're checking against MISRA-C:2004) or mandatory & required rules (if you're checking against MISRA-C:2012). I've found that when compiling with -pedantic -ansi switched on, you already catch many things which are also either required / advised by MISRA. The best would be to ignore the advised directives for now.

podhrmic, 2016-07-13:
I wrote down first few rules from MISRA-C standard: https://wiki.paparazziuav.org/wiki/Developer_Guide#MISRA-C_Rules

It is kind of grey territory, because there explanations are taken from _Understand for C_, and they take it from MISRA-C standard (can be purchased for ~20$). So I am not sure if actually putting it on a public wiki is legit? @flixr @deonblaauw ?

flixr, 2016-07-13:
No idea if you can restate them or not...
In general I think most people won't read all those anyway... so personally I would prefer an updated/expanded http://docs.paparazziuav.org/latest/stylec.html that just covers the most important guidelines.

podhrmic, 2016-07-13:
How can I locally compile the doxygen file? So I can check it for syntax/formatting errors before committing?

flixr, 2016-07-13:
`make dox`

podhrmic, 2016-08-10:
This snipped should replace the MISRA page in wiki (saving here for reference as I still don't have access to the wiki):
==[[Coding Standards]]==
JPL (Jet Propulsion Laboratory) released their own coding standards (can be downloaded for free here: [http://lars-lab.jpl.nasa.gov/JPL_Coding_Standard_C.pdf JPL C Coding Standards and Guidelines]). They include MISRA-C rules and directives, but add their own rules on top of that. They also specify different levels of compliance (LOC) because not always it is necessary to comply with all of the rules. Anyway, they are making things that go to space, so we can definitely learn something about good coding practices from them.

podhrmic, 2018-04-14:
I don't think full MISRA compliance is feasible in the short term. Closing as it is no longer relevant.

closed 2018-04-14 by podhrmic