paparazzi/paparazzi #1801: radio_control=ppm is not working reliably with apogee_1.0_chibios

martinmm, 2016-07-14:
In recent master (cf74dae) radio_control=ppm is not working reliably with apogee_1.0_chibios. With a 7 channel Futaba receiver the gcs RC indicator goes to NONE with sticks positioned low, recovers in middle position.

The NONE appears at a ppm sync gap value of around 10.7ms and higher. With RC_PPM_TICKS_PER_USEC=6 that would be in the range of a 16 bit saturation. The icuoverflowcb() gets called in that case.

gautierhattenberger, 2016-07-14:
The TIM1 used for ppm input is indeed a 16bits counter, which saturates after 10.9 ms with 6 ticks per us. The number of ticks can be changed in the board.h file, but what would be an acceptable value ?
5 -> max gap 13.1 ms
4 -> 16.4
This will also reduce the accuracy of the pulse measurements. But I guess 4000 or 5000 steps for regular RC is good enough...

martinmm, 2016-07-14:
The period of the Futaba receiver is 22ms. The minimum pulse length is 1ms. With 7 pulses the gap could be up to 15 ms in theory. Guess there are no receivers with less than 7 channels?

SBUS uses a sweep of 2048 steps, PPM does have about 1000. If the tick frequency is at least 2x the ppm frequency we should be fine, right?

gautierhattenberger, 2016-07-14:
If we set to 2 ticks per us, it makes a max period of 32 ms and 2000 steps. So it should work in all cases. Do you agree ?

martinmm, 2016-07-14:
Sounds good!

gautierhattenberger, 2016-07-14:
closed with https://github.com/paparazzi/paparazzi/commit/d1969c813445d369828d4a0494b52e9407bc1f7d

closed 2016-07-14 by gautierhattenberger

martinmm, 2016-07-15:
Works. Thanks Gautier!