paparazzi/paparazzi #1819: Guided mode without RC

gautierhattenberger, 2016-07-20:
It seems that it is not possible to fly in guided mode without RC, or at least it is not possible to start a mission in NAV, then switch to GUIDED and back to NAV (using `autopilot_mode_auto2` setting) if the RC is not turned ON.
Since GUIDED is almost a NAV mode, it should be possible to do that.

flixr, 2016-07-20:
You mean because RC_LOST_MODE is triggered?
I guess a first simple solution would then be to also check for GUIDED mode in failsafe_check, but we should probably be a bit more clever and still trigger that if you loose datalink and RC (assuming you use GUIDED mode over datalink) or something like that...

gautierhattenberger, 2016-07-20:
It's not the only issue. `autopilot_mode_auto2` is only checked and used in the rc frame handler, so it is useless without RC. only changing failsafe check would allow a full flight in guided mode, but not switching between nav and guided.

flixr, 2016-07-21:
Right... but `autopilot_mode_auto2` only makes sense with RC as it simply means which mode the AUTO2 RC 3way-switch position stands for.
MANUAL, AUTO1 and AUTO2 have no real meaning without the RC...

We could of course add a settings handler for `autopilot_mode_auto2` that actually applies the changed mode even if you don't have RC. E.g. if the current autopilot mode equal the old `autopilot_mode_auto2`, then set the autopilot mode to the new AUTO2 mode...
But to me this feels like a hack...

So IMHO it would make more sense to allow setting `autopilot_mode` to GUIDED and NAV directly via settings as well. Would also need an extra settings handler that checks if you can actually go into GUIDED or NAV.

Also if we allow this, the HOME mode checks should maybe also apply for GUIDED mode?

gautierhattenberger, 2016-07-22:
Hum, good point, I guess you're right with a handler to prevent illegal order:
- kill/home/failsafe always accessible
- if RC ON, don't change modes (use RC switch) unless for safety modes
- if RC OFF, allow to change nav-like modes (nav, guided, hover_z_hold ?, module ?) + safety modes

Do you see something else ?

flixr, 2016-07-22:
basically sounds good to me (allowing to switch to hover_z_hold probably doesn't make any sense if you don't have RC though)

flixr, 2016-09-19:
@gautierhattenberger is there still something to do here, or is it completely solved with #1852?

gautierhattenberger, 2016-09-19:
no, it's done

closed 2016-09-19 by gautierhattenberger