paparazzi/paparazzi #1823: GCS Crashes on Ubuntu 16.04 when using papgets in layout file

deonblaauw, 2016-07-22:
GCS segmentation fault occurs when there are papgets in the GCS layout file. This happens on Ubuntu 16.04. The crash occurs immediately when telemetry is received. When no telemetry is received, GCS does not crash, but papgets also do not display.

dewagter, 2016-07-22:
I confirm I have had this too (only on ubuntu16): layout horizontal.xml
works but bottom_settings.XML gives a segfault in the GCS.

flixr, 2016-07-22:
Hm... very weird.
Works fine with e.g. papgets.xml on our 16.04 pprz-dev Docker image, but then it's using the X-Server of my Ubuntu 14.04 host...

Could someone with native 16.04 maybe try to dig deeper? Run it in gdb or so...

dewagter, 2016-07-23:
@openuas could you try it on the field laptop with gdb?

OpenUAS, 2016-08-03:
Yes, even on development laptop ...so I have the  .crash file and I will try something like:
https://wiki.ubuntu.com/DebuggingProgramCrash

GCS does however it doesshow papgets for a second or so
[image]
 but I have the hunch that after TELEMETRY comes in play the GCS crashes...

...ongoing , everyone is frree to join ;) with their debugging since this week there are some other urgent task..

commit 08cd0ff5 (2016-08-16): [ocaml] fix GCS crash when using papgets

was due to the use of 'magic' instead of the proper function
close #1823
closed 2016-08-16 by gautierhattenberger