paparazzi/paparazzi #1833: GCS spams unfinished sentences

EwoudSmeur, 2016-08-02:
After the successful resolution of https://github.com/paparazzi/paparazzi/issues/1046, I am back with a new issue. The GCS still does not finish some sentences because it starts a new one and can in some cases start to spam messages. This can happen for instance when you turn off a drone, if I remember correctly it can then start to alternate between "all links lost" and "approaching".

Now, the approaching message is actually the one I feel is the least interesting message of them all, though you hear it most often.

gautierhattenberger, 2016-08-02:
I was also thinking of removing it, but a nicer option I think would be to trigger the "approaching" only if you actually flew a minimum distance (it is only useful when flying long segment after all).

dewagter, 2016-08-02:
It often alternates between for instance "Battery Low" and another message.

-Christophe

EwoudSmeur, 2016-08-10:
That would be useful. Or maybe an option to disable the approaching message would be nice? I was not completely aware of the function of the approaching message, initially I thought it was approaching HOME. Now I understand it is to notify the operator that the drone is about to finish a segment. However, if I check what it does in the simulator for rotorcraft it says approaching every time it leaves for a new waypoint...

Perhaps stopping all messages for drones that lost link (or everything except link loss) would be nice. Now sometimes when you turn off a drone you get spammed to death, while there is little use of messages when there is no link

PR #1842 (2016-08-17): [gcs] prevent speech spam with 'approaching'
only triggered when close enough but not too much, and flying a long segment
fix #1833
commit e1ff1dc8 (2016-08-17): [gcs] prevent speech spam with 'approaching'

only triggered when close enough but not too much, and flying a long segment
fix #1833
closed 2016-08-17 by gautierhattenberger