paparazzi/paparazzi #1840: undeclared baud rates in cserial.h on mac OS

gunderscoreb, 2016-08-10:
(El Capitan)
Maybe this could help? https://developer.apple.com/library/mac/samplecode/SerialPortSample/Listings/SerialPortSample_SerialPortSample_c.html

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C sw/lib/ocaml
OC cserial.c
cserial.c:58:5: error: use of undeclared identifier 'B921600'
    B921600,
    ^
cserial.c:59:5: error: use of undeclared identifier 'B1500000'
    B1500000,
    ^
cserial.c:60:5: error: use of undeclared identifier 'B3000000'
    B3000000 };
    ^
3 errors generated.
make[1]: *** [cserial.o] Error 2
```

podhrmic, 2016-08-10:
Hmm, and one would hope that Mac OS has the same serial drivers as Linux...see https://github.com/tomszilagyi/gen_serial/issues/1  @flixr should we guard this with ifdefs or is there some smarter way to enable this on Mac?

flixr, 2016-08-10:
I think guarding with ifdefs is the only way that really works...

podhrmic, 2016-08-10:
Fixed with 8bff7aa9e4f3b3e080a108e4510781cb7ea738bb 
@gunderscoreb can you test pls?

gunderscoreb, 2016-08-10:
Yeah, it does the trick :) Thx @podhrmic!
(I wish I had the knowledge to help and enable those higher baud rates)

flixr, 2016-08-10:
@podhrmic I think the case that you request a higher baudrate from ocaml than osx supports still needs to be handled... it will currently index the array out of bounds

podhrmic, 2016-08-10:
Does OCAML have something like #ifdefs ? That would do the trick...

flixr, 2016-08-10:
We would have to run it through the camlp4 preprocessor or do it at runtime...
But I think it would be easier (and safer) to check this in the C code to make sure we don't index out of bounds and return -1 or so on failure...

gunderscoreb, 2016-08-10:
Maybe there is a way to enable those baud rates in osX? Might the link I shared before be of any help?

https://developer.apple.com/library/mac/samplecode/SerialPortSample/Listings/SerialPortSample_SerialPortSample_c.html

```
  // The IOSSIOSPEED ioctl can be used to set arbitrary baud rates
    // other than those specified by POSIX. The driver for the underlying serial hardware
    // ultimately determines which baud rates can be used. This ioctl sets both the input
    // and output speed.
    
    speed_t speed = 14400; // Set 14400 baud
    if (ioctl(fileDescriptor, IOSSIOSPEED, &speed) == -1) {
        printf("Error calling ioctl(..., IOSSIOSPEED, ...) %s - %s(%d).\n",
               bsdPath, strerror(errno), errno);
    }
```

gunderscoreb, 2016-08-10:
(I'm aware that of the 3 of us, I'm the only one with a mac. I don't quite understand the snippet above but if you do and can explain it to me, I'm happy to chip in).

closed 2016-08-10 by gunderscoreb

flixr, 2016-08-11:
Thx, LGTM