paparazzi/paparazzi #1843: I2C ublox module requires a UART port.

gunderscoreb, 2016-08-18:
gps_i2c_ubx module depends on the UART-based gpx_ublox module and as such, requires a UART port to be defined.

flixr, 2016-08-18:
So this was added by @masierra with #1580 

At a quick glance it looks like it should still work with the following configuration:
```
<module name="gps" type="ublox">
  <configure name="UBX_GPS_PORT" value="gps_i2c"/>
</module>
<module name="gps" type="ubx_i2c">
  <configure name="GPS_UBX_I2C_DEV" value="i2c1"/>
</module>
```

@gunderscoreb can you plz test if that works?

But this is definitely not particularly nice to use... as already discussed in #1580 it would make sense to "merge" the i2c support into the normal gps_ublox module and directly use `<configure name="UBX_GPS_PORT" value="i2c1"/>`.
Also as @gautierhattenberger mentioned the ublox code should be refactored to have a generic part and deal with the "link device" separately.
Anyone up for this (at least the user facing part of having one module with easier configuration)? @masierra ?

masierra, 2016-08-18:
What do you think about declaring  `<module name="gps"    type="ubx_i2c">` and then autoloading gps ublox in the xml?

gunderscoreb, 2016-08-19:
@esden: will do when I get back to my autopilot either over the weekend or Monday.

flixr, 2016-08-19:
@masierra that sounds like a good enough interim solution

gunderscoreb, 2016-08-23:
Thx guys for working on fixing this! I haven't been able to have it working yet but might still have things to fix on the hardware side. Will share updates.

gunderscoreb, 2016-08-24:
@flixr, @masierra: This may now be becoming off-topic, but I'm still not successful at getting GPS data via I2C.
I'm using elle0 and this GPS: https://store.uputronics.com/index.php?route=product/product&path=60_64&product_id=83
My airframe is configured as follows:

    <define name="USE_I2C2" />
    <define name="I2C2_CLOCK_SPEED" value="50000"/>
    
    <module name="gps" type="ubx_i2c">
      <configure name="GPS_UBX_I2C_DEV" value="i2c2"/>
    </module>

The code compiles and load fine, but communication with the GPS doesn't seem to happen. (the GPS message has only zeros).
Scoping the i2c channels, the lines go down once at startup, for about 3.25 us, and then remain high.

Otherwise, I believe that my GPS and i2c port are both working, respectively.

Any clues on what may be going on?

gunderscoreb, 2016-08-25:
Ok, I finally got it to work Thanks guys. Another module that may deserve to be autoloaded is ucenter (the `gps_ubx_i2c_ucenter_done` flag needs to be set at startup).

flixr, 2016-08-30:
Um... it shouldn't depend on ucenter...
@masierra would be great if you could look into that..

flixr, 2016-09-18:
@masierra ping... does it work without ucenter?

flixr, 2016-09-26:
any updates here?

gunderscoreb, 2016-09-26:
As present, the import of GPS messages only take effect after the gps_i2c_begin() function is called.
This function is called in the ucenter module.

The shortest fix would probably be to modify line 98 with
```
#ifndef GPS_UBX_UCENTER_H
gps_ubx_i2c_ucenter_done = TRUE;
#else
gps_ubx_i2c_ucenter_done = FALSE;
#endif
```
(and renaming gps_ubx_i2c_ucenter_done to something more appropriate)

but imho it's probably simpler to autoload center.

flixr, 2016-09-26:
Thanks.
Requiring ucenter is not so nice, as then you can't disable the autoconfiguration...
Should be fixed with https://github.com/paparazzi/paparazzi/commit/ef0c774a14cbf34deb15dd27f4b6cb1dcda8e8f3, can you plz test latest master and report back?

gunderscoreb, 2016-09-30:
seems to work.

flixr, 2016-09-30:
:-) thanks for testing

closed 2016-09-30 by flixr