paparazzi/paparazzi #1854: RTP video stream is unstable

wilcoschoneveld, 2016-09-12:
I would like to address the RTP video stream in order to improve the computer vision development and debugging experience. Currently it is lacking in desired performance:
- The first frame takes a while to load (routinely >5 seconds)
- Frame rate is very unstable
- In-stream freezes of ~1-5 seconds are very common
- Sometimes the entire stream needs to be re-loaded

It's very common to have the following console output when trying to view the video stream;
```
[sdp @ 0xa32900] Received packet without a start chunk; dropping frame.
[sdp @ 0xa32900] Received packet without a start chunk; dropping frame.
[sdp @ 0xa32900] Missing packets; dropping frame.
[sdp @ 0xa32900] Missing packets; dropping frame.
[sdp @ 0xa32900] Missing packets; dropping frame.
[sdp @ 0xa32900] Missing packets; dropping frame.
[sdp @ 0xa32900] Missing packets; dropping frame.
..
..
```

I have zero to little knowledge about the implementation and history of viewvideo and can only guess if there is room for improvement. It would be great if I could get some input on this.

@gautierhattenberger @fvantienen

fvantienen, 2016-09-12:
I have a lot of improvement ready somewhere.. you can contact me if needed ;)
But the best performance improvements I got from changing to UNICAST instead of BROADCAST the UDP packets.

wilcoschoneveld, 2016-09-12:
Awesome! Do you have a branch with updated code? I'm not sure how to set UNICAST, just setting BROADCAST=FALSE isn't correct I suppose

wilcoschoneveld, 2016-09-16:
I got it running and indeed UNICAST seems to work very well; no more console warnings and a stable stream. Thanks!

conf/modules/video_rtp_stream.xml:48
```
      VIEWVIDEO_HOST        ?= 192.168.42.68   <-- viewer ip on bebop wifi
      VIEWVIDEO_PORT_OUT    ?= 5000
      VIEWVIDEO_BROADCAST   ?= FALSE
      VIEWVIDEO_USE_NETCAT  ?= FALSE
```

sw/tools/rtp_viewer/rtp_stream.sdp
```
v=0
m=video 5000 RTP/AVP 26
c=IN IP4 192.168.42.1
```

Right now I have to set it in the raw section of the module file because setting it in the airframe will give a bunch of redefine errors. I'm not too familiar with the build system, how could I fix this?

flixr, 2016-09-17:
@wilcoschoneveld does the doc section in https://github.com/paparazzi/paparazzi/blob/master/conf/modules/video_rtp_stream.xml#L14 answer your question on how to configure it from your airframe file?
Also see the docs generated from that xml file: http://docs.paparazziuav.org/latest/module__video_rtp_stream.html

wilcoschoneveld, 2016-09-21:
@flixr Yes perfect! Thanks a lot for pointing it out to me.

closed 2016-09-21 by wilcoschoneveld