paparazzi/paparazzi #1857: Adapted Throttle causes drop in guidance height

knmcguire, 2016-09-19:
Hi, I have an issue with the vertical guidance with ADAPT_THROTTLE_ENABLED on true. 

If I switch from AP_MODE_ATTITUDE_RC_CLIMB to AP_MODE_HOVER_Z_HOLD with the Bebop 1 at 1-2 meters, it drops to the ground and slowly rises to the correct height. However I want it to stay at that height from the beginning 

I had the same problem with AP_MODE_ATTITUDE_DIRECT  to hover, but that is because it didn't adapt the intergral factor by the adapt throttle. However, with RC_CLIMB, it is already using the adapt_throttle in guidance... 

Anybody have any thoughts on how to fix this? Thanks in advance.

agressiva, 2016-09-19:
Some time i have the same problem when using adaptive_throttle
With adaptive disabled the quad never drop altitude.

flixr, 2016-09-19:
Do you have any logs that contain this? It's kind of hard to guess where the issue is without any actual data...

We should probably not run the adaptive throttle estimator if we are in DIRECT mode... but it *should* work fine when switching from any other mode that internally runs the hover loop.

flixr, 2016-09-19:
Also if the acceleration is over a threshold (`GUIDANCE_V_ADAPT_MAX_ACCEL`) the estimator is not updated... and since the acceleration in the state interface is fed by the INS, it also depends what INS you use and how "clean" the estimated current acceleration is.

kirkscheper, 2016-10-31:
For us, this is 'solved' with 1917...

commit 6ca4ce4e (2016-12-17): [rotorcraft] only run adaptive throttle estimation if desired zd was updated

Only run adaptive throttle estimation if we are in flight and
the desired vertical velocity (zd) was updated (i.e. we ran hover_loop before).
This means that the estimation is not updated when using direct throttle commands.

This should fix #1857
PR #1972 (2016-12-20): only run adaptive throttle estimation if desired zd was updated
Only run adaptive throttle estimation if we are in flight and
the desired vertical velocity (zd) was updated (i.e. we ran hover_loop before).
This means that the estimation is not updated when using direct throttle commands.

This should fix #1857
commit fa7a5801 (2016-12-20): [rotorcraft] only run adaptive throttle estimation if desired zd was updated (#1972)

Only run adaptive throttle estimation if we are in flight and
the desired vertical velocity (zd) was updated (i.e. we ran hover_loop before).
This means that the estimation is not updated when using direct throttle commands.

This should fix #1857
closed 2016-12-20 by flixr