paparazzi/paparazzi #1899: Video thread(s) not running in latest master

wilcoschoneveld, 2016-10-12:
The video_thread_function (sw/airborne/modules/computer_vision/video_thread.c:83) does not seem to start in the latest master. It still works as of commit fe5500811fa46d51c427cbc7cc83d5e0d1a973e2. 

I looked into the commits but couldn't find a hint of where it broke. Could this be because of the refactored telemetry modules? @gautierhattenberger 

Thanks!

flixr, 2016-10-12:
You did run a toplevel `make clean && make` or at least a `make` right?

wilcoschoneveld, 2016-10-12:
Yes, I did

kirkscheper, 2016-10-13:
This is confirmed on a different bebop using a different pc. Both cameras don't startup for some reason.

flixr, 2016-10-13:
Please always specify exactly which version you tested...
Can you try to narrow down on the commit that broke this? E.g. try right before and after #1835 was merged with 4316871b6324276d2c3128dbf09d7664beea9142

wilcoschoneveld, 2016-10-13:
I tested master at commit 183ac5cfe627311a7a78ac277d1cf62a529559d2. I understand that my issue description is not really extensive and this is because I have not investigated this issue as much; we are really strapped for time due to IMAV coming up next week.

flixr, 2016-10-28:
Closed with 44f15c363d09a066c04c68b6848cc67aa4c732da

closed 2016-10-28 by flixr