paparazzi/paparazzi #1923: Modules and main_periodic don't always have the same order

EwoudSmeur, 2016-11-02:
When using the logger_file module on the bebop1 at the same speed as main_periodic, I find that at irregular moments I get 2 times the same values in the log file. When I put a counter in main_periodic and log that, I see that it sometimes records the same value twice, after which it skips one.

This lead me to the hypothesis that main_periodic and modules_periodic_task sometimes switch order. It seems to me that this should not be the case, but I am not sure.

flixr, 2016-11-02:
MODULE_FREQUENCY is configured separately from PERIODIC_FREQUENCY. How do you have those set?
If you enable the "print config at build time" checkbox, you can see it in the compile output.

fvantienen, 2016-11-02:
This is theoretically possible but the chances are extremely low. Since in between the check for the periodic timer of the main periodic and modules periodic the module timer can be set. This is possible because the periodic timer runs in another thread.

Because it happens quite frequently I don'the think that is the exact problem. But I think it has to do something with multithreading.

flixr, 2016-11-02:
Yes, this can happen... although strictly speaking the "order" doesn't change...
E.g. in rotorcraft main you have:
```
  if (sys_time_check_and_ack_timer(main_periodic_tid)) {
    main_periodic();
  }
  if (sys_time_check_and_ack_timer(modules_tid)) {
    modules_periodic_task();
  }
```
and while this is executed in the same _thread_, the sys time advances between the two calls and it can be the case that when checking the main_periodic the timer is not yet elapsed, but the modules timer is already elapsed when it is checked after that (even if they have the same frequency).
Also the timers start running when they are registered... so if they are not registered at exactly the same time (nearly impossible) the won't elapse at exactly the same time either...

flixr, 2016-12-16:
@EwoudSmeur what is your exact use case and expected outcome here?
Maybe we could change the code to only add a separate modules timer if MODULE_FREQUENCY is different from PERIODIC_FREQUENCY and only use the `main_periodic_tid` if they are the same...

EwoudSmeur, 2016-12-16:
That sounds like a good solution! 
Right now I am just concerned with the logger, I would like that every time the control loop runs I can log the important values. If it now and then it skips a value, I may need to do some extra smoothing etc to present the results nicely in a paper.

gautierhattenberger, 2016-12-16:
I seems a good idea to me as well

PR #1971 (2016-12-16): [rotorcraft] re-use main timer if PERIODIC_FREQUENCY == MODULE_FREQUENCY
Use the main periodc freq timer for modules if the freqs are the same
This is e.g. useful to log the important values every time the control loop runs.

should fix #1923
commit fb4d3f42 (2016-12-16): [rotorcraft] re-use main timer if PERIODIC_FREQUENCY == MODULE_FREQUENCY (#1971)

Use the main periodc freq timer for modules if the freqs are the same
This is e.g. useful to log the important values every time the control loop runs.

should fix #1923
closed 2016-12-16 by gautierhattenberger