paparazzi/paparazzi #1924: Calling AbiBindMsgRPM twice leads to unresponsive system

EwoudSmeur, 2016-11-02:
Noticed on a bebop1. After telnetting I could see that a printf in the callback gave a lot of output, but I did get any telemetry. When I make sure AbiBindMsgRPM is called only once I have no problem.

I know the function is not supposed to be called twice, but I also think it should not make the system unresponsive.

flixr, 2016-11-02:
Hm... I don't see how that could be a problem with ABI itself.
If you call AbiBindMsgX multiple times, it will simply call the callback function multiple times (with the one bound last called first).
You can see that in the generated var/include/abi_messages.h

So I think the cause is somewhere else... maybe depends on what you do inside the callback?

flixr, 2016-11-02:
Or are you re-using the same abi_event for the second bind?
I guess that would create an endless loop as the event will add itself as next in the linked list...

PR #1925 (2016-11-02): [abi] fix endless loop if AbiBindMsgX is called twice for same event
AbiBindMsgX should be called only once per abi_event struct.
Regardless, this tries to make it a bit more robust if you re-use the same abi_event for consecutive AbiBindMsgX calls:
If AbiBindMsgX is called multiple times _consecutively_, the event/callback is now only added if it is different from the last one added...
This should at least fix the endless loop since the event->next can't point to itself anymore...

Should fix #1924 in a rudimentary way... but would still be possible to bind multiple times using the same abi_e [... cut]
commit 34e5f801 (2016-11-02): [abi] fix endless loop if AbiBindMsgX is called twice for same event (#1925)

AbiBindMsgX should be called only once per abi_event struct.
Regardless, this tries to make it a bit more robust if you re-use the same abi_event for consecutive AbiBindMsgX calls:
If AbiBindMsgX is called multiple times _consecutively_, the event/callback is now only added if it is different from the last one added...
This should at least fix the endless loop since the event->next can't point to itself anymore...

Should fix #1924 in a rudimentary way... but would still be possible to bind multiple times using the s [... cut]
closed 2016-11-02 by gautierhattenberger