paparazzi/paparazzi #1936: realtime plotter spams warnings (ubuntu 16.04)

EwoudSmeur, 2016-11-03:
The realtime plotter typically spams this warning for me:
(plotter:31131): GLib-GObject-WARNING **: value "-nan" of type 'gdouble' is invalid or out of range for property 'lower' of type 'gdouble'

Other than that it works fine. But now I can not see anything that happens in paparazzi center...

podhrmic, 2016-11-03:
Same thing here:(

podhrmic, 2016-11-08:
Update - the warning comes from this line in glib: https://github.com/GNOME/glib/blob/master/gobject/gobject.c#L1414

And indeed, this is caused by the ocaml plotter https://github.com/paparazzi/paparazzi/blob/master/sw/logalizer/plotter.ml - it triggers the warning when it a message is not received (lets say telemetry is lost), but time is still going so the plotter gets updated with nan values. Can't find out where it is being called though. Any insights from our Ocaml gurus @flixr @gautierhattenberger ?

podhrmic, 2016-11-08:
Update - it happens also when the same value is updated (lets say you are plotting a battery voltage and it is not changing, but it is being received from a message) -> still the warning...

gautierhattenberger, 2016-11-08:
Hopefully this was the only source of warnings...

commit 1c7d9541 (2016-11-08): [plotter] disable usused std and average computation to prevent glib warning (fix #1936)
closed 2016-11-08 by gautierhattenberger